OpenSprinkler/OpenSprinkler-Firmware #11: Doesn't turn on sprinklers

onionjake, 2015-04-23:
I have the OSpi.  I was able to build this project, start it, browse to the webpage, and log in.  I manually turn on a station, but the sprinklers do not turn on.  If I use the interval_program, the sprinklers turn on so I know it is not a problem with my wiring, the board, or the sprinklers.

I tried both version 2.1.3 and master.  What am I missing?

salbahra, 2015-04-23:
Are you running OpenSprinkler as root (or sudo)? The GPIO control requires root permissions.

onionjake, 2015-04-23:
I start it with `sudo /etc/init.d/OpenSprinkler.sh start`, so yes.

onionjake, 2015-04-23:
If it matters, I have the OSPi and not the OSPi+.

rayshobby, 2015-04-23:
Hi, can you specify what browser you are using? Also, can you try to send an HTTP GET command directly, such as http://devip/cm?pw=xxx&sid=1&en=1&t=60, where as devip is your OSPi's IP address, xxx is your password (it may be md5 hashed, for example, the md5 hash for 'opendoor' is a6d82bced638de3def1e9bbb4983225c)

onionjake, 2015-04-23:
Version 41.0.2272.76 Ubuntu 14.10 (64-bit)

I can see in the logs that the sprinklers are being turned on?  So I don't think it is a browser issue.

```
[99,5,120,1429749552]
[99,4,120,1429749672]
[99,5,13,1429749741]
[99,4,14,1429749953]
[99,5,25,1429750906]
```

I'll work on getting a port opened up.  I'd prefer to send the credentials through email, which should I use?

onionjake, 2015-04-23:
Never mind about the email, I didn't quite understand what you were asking.

This is what I get:
```
{"result":1}
```

Sprinklers still don't turn on.  I enabled DEBUG printing, and this shows up in the output: `[1:1429751412,1429751472]`

rayshobby, 2015-04-23:
We just made a couple of changes to the HTTP header, which may fix the problem. Try to git pull, recompile, reboot and see what happens. Also, try to use Firefox or Safari and see if it's the same behavior.

onionjake, 2015-04-23:
I pulled them, rebuilt, and rebooted.  I tried Firefox as well.  Still no luck. 

I think the web page is working fine, as it indicates that the sprinkler should be on...

[image]

rayshobby, 2015-04-23:
Well, honestly I have no idea, because I cannot reproduce the issue.

rayshobby, 2015-04-23:
You know what, I just realized, you have the older RPi version (rev. 1), and there is pin number difference. Go to defines.h, change the #define PIN_SR_DATA (under #if defined(OSPI) section) from 27 to 21. Then recompile and try it again. Let me know if this fixes the issue

onionjake, 2015-04-24:
Changing the pins worked.

salbahra, 2015-04-28:
Fixed in 7a6fa81527ffb459273a934bf9c9f706c9eaed7e

Thanks for the report!

closed 2015-04-28 by salbahra