OpenSprinkler/OpenSprinkler-Firmware #13: Manual run programs are not logged

denisfrench, 2015-05-03:
I'm a bit surprised programs that have been run manually do not appear in the log.

rayshobby, 2015-05-04:
Manual run program are logged. The issue might be in your microSD card.

closed 2015-05-04 by rayshobby

denisfrench, 2015-05-04:
Okay, likely I'm missing something here.

I thought initiating a manual run by calling manual_start_program() was
using the same method as the Arduino UI.

Calling manual_start_program() and reset_all_stations() is successfully
starting and stopping run-once programs, but no log of the manual runs (or
the station that was interrupted) results.

I can see that station logging occurs when turn_off_station is called, but
not how this happens for the interrupted station when ui_state_machine()
does a manual run, or when a manual stop occurs.

Any pointers greatly appreciated.

Best regards,
Denis.

☎ +618 707 111 65