OpenSprinkler/OpenSprinkler-Firmware #14: [Enhancement] Turn on stations scheduled to be on upon start-up and end of manual runs

denisfrench, 2015-05-03:
I get this is a bit tricky, but it would be helpful if stations scheduled to be on as per their regular program were turned on upon start-up/reboot, and at the completion of manual runs.

salbahra, 2015-05-04:
Ray is going to switch to a station queue system and I think if the queue is saved in EEPROM or another persistent location then it would accomplish this task.

brecht1234, 2015-10-29:
Was this feature included in the 2.1.6 release?

rayshobby, 2016-01-16:
No, this is not included in 2.1.6 release. At the moment it's not easy to resume a running program across reboot, because this requires non-volatile memory (such as FRAM) that can be reliably written many times. Doing so would require a hardware change.

closed 2016-01-16 by rayshobby