OpenSprinkler/OpenSprinkler-Firmware #16: Plugins menu is only available from the home screen

xaque208, 2015-05-05:
At the web interface to configure the OpenSprinkler, the menu for plugins appears only on the home screen. So if I click "log" on top, for example, I do not see the plugins drop down that I see after clicking the home icon.

Running 2.0.3 on a beagle, with a CPU temp of 53C.

salbahra, 2015-05-05:
Apologies but you have the wrong repository. I believe you are looking for: https://github.com/Dan-in-CA/OSPi

Also, Dan's OSPi is now on 2.2 so I recommend upgrading to make sure the issue is still present.

Thanks

closed 2015-05-05 by salbahra

xaque208, 2015-05-06:
Is the OSPi repo you linked the source for the web interface for the OpenSprinkler project?  I'll read the docs on upgrading.

salbahra, 2015-05-07:
The upgrading documentation is here: http://rayshobby.net/mediawiki/index.php/Python_Interval_Program_for_OSPi#Updating_the_program

That package has everything contained in one repository.

The official OpenSprinkler app on all stores support this firmware for the OSPi. It also supports the Unified firmware which is this repository hosting the issue we are discussing.

xaque208, 2015-05-11:
I've upgraded this morning.  Looks much better.  Thanks for the links.  I may consider trying to package the thing for just an `apt-get upgrade` with the right repo in place, and viola.