OpenSprinkler/OpenSprinkler-Firmware #17: Build instructions?

xaque208, 2015-05-11:
I'm looking into helping build out the platform for the opensprinkler project, and I'm looking for build instructions.  Is there a document (or even rough notes) on the steps you take to get the image built for the osbo?  I'd like to try and get the system running on Debian, for religious reasons, and package what isn't packaged.  I'm happy to host a repo and such, but I think packages would make this easier from a configuration management perspective.  I'd like to connect the Beagle to the rest of my Puppet infrastructure, and in the process will likely write a module for management.

It all begins with the instructions of how the Ubuntu beagle system I'm running now got into its current state though.  Anything you can link would be much appreciated.

salbahra, 2015-05-12:
There really isn't anything to it. So long you have Debian installed, just do the following:

1) apt-get install git build-essential
2) git clone https://github.com/opensprinkler/opensprinklergen2
3) cd opensprinklergen2
4) ./build.sh osbo

At this point, you could shutdown and distribute the image, ready to go. Build.sh will install an init.d script and add it to startup (it will prompt you during the script).

closed 2015-05-12 by salbahra

xaque208, 2015-05-12:
Excellent, thank you.