OpenSprinkler/OpenSprinkler-Firmware #21: later compiler/bintools/ support (avr-gcc 4.8.3)

ifreislich, 2015-06-10:
I currently maintain a set of private patches in order to compile the OpenSprinklerGen2 code on arduino-1.0.6 with avr-gcc-4.8.3,  avr-binutils-2.25 and avr-libc-1.8.1.  I'm pretty sure that these are later versions than the bundled items with the Arduino IDE and at some future time will have to be supported.  There are several compile-time issues:

1. prog_char is deprecated and does not name a type.
    prog_uchar does not name a type.
2. Several warnings regarding comparison between signed and unsigned integer expressions.
3. non- const variable declaration with '__attribute__((progmem))'
4. adding '__attribute__((progmem))' to wtopts_name results in the object name '_ZL11wtopts_name' and the linker is unable to resolve the symbol in object files where it was declared with the extern keyword.

I do not yet have a fix for (4), only a work-around that doesn't declare the variable as PROGMEM.

If you are interested and let me know what your toolchain versions are, I'll prepare a  pull request for what I have now excluding (4) and then spend more time working on a proper fix for (4).

kpfleming, 2015-06-10:
You can probably fix item 4 by wrapping the declaration with an ```extern "C" { ... }``` block.

ifreislich, 2015-06-10:
Unless I did it wrong, it changes nothing:
  extern "C" { prog_char wtopts_name[] PROGMEM = WEATHER_OPTS_FILENAME; }
...
[zen] /tmp/build4541738100628912497.tmp/OpenSprinklerGen2 $ /usr/local/avr/bin/nm OpenSprinkler.cpp.o |grep wtop
00000000 r _ZL11wtopts_name

closed 2015-07-29 by ifreislich