OpenSprinkler/OpenSprinkler-Firmware #23: Switch to manual

0-kaladin, 2015-06-19:
Is switch to manual gone?  Can't find it now after switching from OSPi
http://rayshobby.net/opensprinkler/svc-use/svc-web/#manual

salbahra, 2015-06-19:
The idea of switching between manual and programmed mode is gone.

However, you can certainly still manually queue stations from the home page by tapping the station name. It will be added to the queue or immediately start if the queue is empty.

closed 2015-06-19 by salbahra

0-kaladin, 2015-06-19:
Thanks that wasn't clear.  when switching on manual mode it cancelled any running programs.  if a program is running on zone 2 and i just click on zone 5 does it cancel zone 2 and just manually run what was clicked?  Or since you used the word 'queue' does it consider the program all in the queue and then would add my click to happen after the program has executed?  
if i click on 3 stations how long does it run them if they were queued and not executed immediately?  
I guess i was just used to a manual mode that clearly overrode any programs and each station you could define a run time so in manual mode it wouldn't just run forever.

salbahra, 2015-06-19:
Sometimes, it's a lot easier to try then to explain. So to that end, here is the link to the demo of the latest software: http://demo.opensprinkler.com.

When you tap a station, you are prompted to enter a duration which defines how long that station will run. If it is a sequential station, it will wait for it's turn to run for that duration. If it's concurrent, it will instantly turn on for the specified duration.

If you want to stop a program in order to do a manual run, the menu button has a red button called "stop all stations" which will allow you to stop the program and manually run your program. Also, keep in mind run-once program will do this for you automatically. This means when a program is defined in run-once, you will be prompted on submit to stop all running stations, telling you it's going to execute a "stop all stations".

0-kaladin, 2015-06-19:
got it.  thanks!