OpenSprinkler/OpenSprinkler-Firmware #31: Please support listening on inet6 when available in addition to inet4

xaque208, 2016-05-18:
Hello, I would appreciate not having to use ipv4 for this service only.  I don't know what consequences this will mean for the mobile applications, but availability over v6 would certainly be handy, so that I could stop allocating a v4 DNS A record for my sprinkler.  I realize I may be a minority in this deployment, but if it doesn't complicate the code much, please.

kpfleming, 2016-05-18:
I'd prefer to use IPv6 as well, and I'm not concerned about the mobile app (Android) because my Android device also has IPv6 from my home network.

futzle, 2016-10-18:
I am working on dual-stack listening for ospi here: https://github.com/futzle/OpenSprinkler-Firmware/commit/3092210f03460282d33e1ccec4ef61b469b6cfd0

dafisher59, 2017-02-12:
I would also like to have IPv6 support.  My ISP sometimes changes the v4 address assigned to my home router, which then breaks my external access until I update my bookmarks and app configurations.  Since the v6 address does not get changed, having a v6 address for the controller would solve that problem and eliminate the need for port forwarding.

K6TD, 2017-02-12:
Use homedns.org and a script to update the hostname to open address. 

KR

futzle, 2017-07-02:
Submitted pull request #49.

xaque208, 2017-07-05:
Yay!

rayshobby, 2017-08-02:
The pull request has been merged. Thanks.

closed 2017-08-02 by rayshobby

dafisher59, 2017-08-02:
Does this apply to all versions of OpenSprinkler, or just to the OSPi version?  I currently have Version 2.3 – AC, and need IPv6 support.

Doug

######################################################################################
Doug Fisher       N7LTW
Software Development Engineer
R & D IT
DXC Technology

douglas.fisher@dxc.com<mailto:douglas.fisher@dxc.com>

Phone: + 1 970-898-9551
DXC Technology
Fort Collins, CO
USA

Scheduled time away:
      FTO:
      Travel:

Closed #31<https://github.com/OpenSprinkler/OpenSprinkler-Firmware/issues/31>.

rayshobby, 2017-08-09:
No, this is only for Linux-based OpenSprinkler like OpenSprinkler Pi. OpenSprinkler 2.3 uses embedded server and I don't think its library currently supports IPV6 at all.