OpenSprinkler/OpenSprinkler-Firmware #50: Docker container for OpenSprinkler on RPi

JohnStrunk, 2017-07-15:
I've been running OSPi using docker for a while, and I'm interested in sending a PR to get the Dockerfile into the main repo.

I'd like to get the log dir and NVM files into a proper docker volume. Currently, I have to do some symlink ugliness in the docker image because OpenSprinkler insists on writing its data files into the same directory as the executable due to get_runtime_path() and get_filename_fullpath().

My current idea is to get rid of the get_filename_fullpath calls and just do a chdir(datadir) in main(). This would allow a datadir to be passed in on the command-line or by default, have datadir=get_runtime_path().

Thoughts?

kpfleming, 2017-07-16:
This sounds like a good idea, even for non-Docker usage.

xaque208, 2017-07-19:
I'm on beagle.

frob, 2018-05-25:
I would use this.

Cabalist, 2018-07-27:
I would also use this.   It would be great to see this merged.

xaque208, 2018-07-30:
I've since purchased the new esp8266 platform and haven't looked back.    I'm still not sure the utility of this, since you'll need to interface with the GPIO, and passing that through the container, while perhaps possible, might be fragile.

closed 2024-05-12 by rayshobby