OpenSprinkler/OpenSprinkler-Firmware #55: No need to wait after Wire.requestFrom()

Koepel, 2017-08-13:
In the file "OpenSprinkler.cpp" there is a while-loop after Wire.requestFrom() to wait for something in two places, but there is nothing to wait for. Those while-loops can be removed.

When the Wire.request() returns, the I2C transaction has completely finished and the received data is waiting in a buffer in the Wire library.

rayshobby, 2017-10-09:
This is based on the description here:
https://www.arduino.cc/en/Reference/WireAvailable

Koepel, 2017-10-10:
The official description of Wire.available() is not very clear. I was better if "_This should be called..._" was replaced by "_This could be called..._". However, they mean the Wire.available() can be used to read the data, since Wire.available() returns the number of data that is still in the buffer. 
The waiting "`while (!Wire.available());`" is never mentioned in the Arduino documentation. That is something that is made up.

If that waiting was needed, then the DS1307RTC.cpp in your project would be wrong.

The Wire.requestFrom() returns the (total and final) number of bytes that are received in that I2C transaction. It is blocking. It is not a discussion.

I see that I posted as well in that link to the forum. Well, anyone can say a lot of things. I suggest to have a look at Nick Gammon's page about I2C: [www.gammon.com.au/i2c](http://www.gammon.com.au/i2c)

When you have investigated this, I would like to go one step further. If the Wire.requestFrom() detects a bus problem (not all bus problems are detected) then it would probably return zero and Wire.available() returns zero. With that while-loop to wait, you would halt the sketch.

rayshobby, 2018-05-28:
This has been marked for deletion in the next firmware. You are right that requestFrom itself returns number of bytes read, so the condition should be based on that and there is no need to wait for available() with a while loop.

closed 2018-05-28 by rayshobby

Koepel, 2018-05-29:
Meanwhile I have made a wiki with common mistakes, explanation of the functions and if checking for bus errors is needed in my opinion: [How-to-use-the-Arduino-Wire-library](https://github.com/Koepel/How-to-use-the-Arduino-Wire-library/wiki). Perhaps it can be useful.