OpenSprinkler/OpenSprinkler-Firmware #56: Problems binding to 8080

bikehead, 2017-09-05:
I have been running OSPI from a standard SD image for about 6 months.  Recently I did  a git pull and rebuilt the firmware.  It built fine but I lost my UI at 8080.  After some poking around I discovered that etherport.cpp is trying to use the IPV6 stack to open the port and this is failing.  I moved etherport.cpp one commit back and the UI returned.

So an upgrade to IPv6 on the standard SD image for OSPI doesn't work for me.

0-kaladin, 2017-09-05:
Weird same for me.  Thanks for the workaround.

kpfleming, 2017-09-05:
This probably means that the kernel on the standard image doesn't have IPv6 enabled by default. It could be built as a module, or not available at all.

If you can ```modprobe ipv6``` successfully, then it's included as a module, and you'd only need to add ```ipv6``` to ```/etc/modules``` to have it loaded on each boot.

pichirichi, 2017-09-12:
Same here, thanks for the workaround
> git checkout 8deb873 etherport.cpp
> sudo ./build.sh ospi
> sudo /etc/init.d/OpenSprinkler.sh restart

Worked like a charm

zirkleta, 2017-10-09:
ipv6 to /ect/modules did not resolve for me.  The rolling up the previous etherport.cpp and rebuilding worked like a charm.

rayshobby, 2017-10-09:
The pre-configured SD card image was build on a much older version of Raspbian and it doesn't support dual IPv4 and IPv6, while the latest firmware, in order to introduce support of IPv6, require that dual support. You should start with the latest Raspbian image and then install the firmware. That should solve the issue. We've now changed all instructions to retire the pre-configured SD card image.

closed 2017-10-09 by rayshobby