OpenSprinkler/OpenSprinkler-Firmware #58: How to indicate error when switching HTTP station?

whmountains, 2017-10-04:
First of all, thanks for creating OpenSprinkler.  The HTTP valve interface makes it very easy to extend it for other applications.

I'm writing a Node.js server that's designed to receive commands from the OpenSprinkler web UI and send them to a remote sprinkler manager over an RF link.  It's working great, but sometimes the remote device is unreachable and I want that to be reflected in the OpenSprinkler UI.

I've tried returning HTTP error codes, but it appears that OpenSprinkler doesn't check the response code.  Is there a workaround for this, or will I have to implement it myself?

whmountains, 2017-10-04:
I've successfully implemented the HTTP response code parsing, but now I'm at a loss as to how to what to do with it.

Any suggestions for how to send the error back to the UI?

rmloeb, 2019-09-28:
I have a similar problem using OpenSprinkler to control an OSBee.  Sometimes the command to turn a zone on or off (latching valve) apparently isn't received, so nothing happens.  There's no way to tell this occurs.  OS thinks the zone state changes and the UI shows that.  I appreciate the complexity of implementation because the zone command is issued from an event, not the UI.  Would it be practical to write something to the log?

rayshobby, 2021-02-03:
The firmware already has a 'special station auto refresh' feature that repeatedly refreshed special (including http and remote) zones periodically to make sure they are in sync. As in my above comment, implementing error feedback is not an easy task and so the work-around is to use periodic refreshing to address this issue in this one-way only communication.

closed 2021-02-03 by rayshobby

whmountains, 2021-02-03:
Wow, this brings back memories!

Since then I started my own irrigation system project powered by LoRa wireless, got acquired by [Farmers Friend](https://www.farmersfriend.com/), and am working to launch a commercial product in the next couple of years.

The HTTP station feature of OpenSprinkler was AWESOME for putting together an initial demo of the technology.  Thank you so much for your work here.