OpenSprinkler/OpenSprinkler-Firmware #59: Hard coding the ssid and password of the router in firmware

SUNSHINE99, 2017-11-29:
Hi,

My project requires that whenever we do device reset or firmware update, we get particular SSID instead of 192.168.4.1. 

Would you please suggest some workaround for the same if any?

Thanks.
Ravi Verma

SUNSHINE99, 2017-11-29:
I mean when we start the device after reset the device is getting started in "AP MODE". When we connect to the router then the devices goes to client mode. 

Is there any workaround where we can hard code the ssid and password somewhere is the firmware so that the device gets connected to same router automatically without needing third device for the same purpose.

Thanks

targetdrone, 2017-11-29:
OpenSprinkler doesn’t have a WiFi interface. It has only Ethernet. You may be confusing this with the network bridge you are using to connect to your WiFi access point.

joeinaustin, 2018-01-19:
targetdrone-you must not be familiar with the newer hardware.  The latest designs are exactly opposite of what you describe.  
SUNSHINE99-It is possible to hardcode your AP info with custom firmware.  But you would be running non-standard code and you may give up any recovery options should your AP fail.  Is that acceptable?

rayshobby, 2018-02-17:
Since we sell the controller to end users who are not developers, we can't hardcode SSID and password. If you want to hardcode that, you can easily modify the firmware to do so.

closed 2018-02-17 by rayshobby