OpenSprinkler/OpenSprinkler-Firmware #65: Build error..

Neomancer86, 2018-05-26:
On a Raspberry Pi 3... getting 

pi@OpenSprinkler:/usr/local/OpenSprinklerGen2 $ sudo ./build.sh ospi
Building OpenSprinkler...
gpio.cpp: In function ‘void attachInterrupt(int, const char*, void (*)())’:
gpio.cpp:411:14: error: ‘delay’ was not declared in this scope
       delay(1) ;
              ^

Neomancer86, 2018-05-26:
i believe this is related to rays commit: 

https://github.com/OpenSprinkler/OpenSprinkler-Firmware/pull/62/commits/f516d9bb59b6604e43ea363e5357f9079bc7432a

rayshobby, 2018-05-27:
The compilation error has been resolved.

closed 2018-05-27 by rayshobby

Neomancer86, 2018-05-27:
yes thanks. although i think there is another issue now related to that -  discussion on the forums regarding the configuration can't be edited.

rayshobby, 2018-05-27:
See my comment: https://github.com/OpenSprinkler/OpenSprinkler-Firmware/issues/67

Neomancer86, 2018-05-27:
perfect. thanks!