OpenSprinkler/OpenSprinkler-Firmware #71: Run once program not respecting weather adjustment

HoECoder, 2018-06-06:
I haven't had a chance to walk through the code on this and do a verification, but it seems like run once programs, when using an existing program, seem to not take the weather adjustment into account.

If this was an intentional design decision, that's OK, but it almost seems like a bug otherwise.

rayshobby, 2018-06-08:
This is intentional: run-once program is assumed to be a one-time manual program. As such, the water time should be specified by the user and is not affected by watering percentage. If you have an existing program, you can run the program manually from the Program page, and it does give you the option to apply watering percentage or ignore it:
https://github.com/OpenSprinkler/OpenSprinkler-Firmware/blob/master/server.cpp#L760

closed 2018-06-08 by rayshobby

HoECoder, 2018-06-08:
Thanks for that clarification! I suspected the case but was not sure. And I didn't know about the feature from the program page, so that's what I'll use in the future.