OpenSprinkler/OpenSprinkler-Firmware #80: Some HTTP stations continue to send 'off' every minute

bretskee, 2019-01-01:
Firmware Version: 1.8.2 (2)
Hardware Version: OSPi

I use OpenSprinkler as a scheduler to switch lights etc around my home as well as the sprinklers. I use HTTP stations to send on and off messages to node-red. I have noticed that some HTTP stations (not all and I have not determined why) continue to send 'off' messages to node-red every minute. This is not ideal given I also have manual switches on these lights. When I switch them on manually, OSPI kindly switches them off again.

rayshobby, 2019-01-01:
Go to Edit Options -> Advanced -> turn off the 'Special Station Auto-Refresh' flag.

closed 2019-01-01 by rayshobby

bretskee, 2019-01-02:
Of course thanks, make sense. What does not make sense however is that I have 4 HTTP stations and only two of them were sending regular 'off' messages. I have confirmed that unchecking the setting does stop all messages. But the inconsistency question still remains.