OpenSprinkler/OpenSprinkler-Firmware #81: Not all HTTP stations send 'off' when Special Station Auto-Refresh is set

bretskee, 2019-01-04:
Firmware Version: 1.8.2 (2)
Hardware Version: OSPi

I have 8 consecutive stations set up as HTTP stations, 17 thru 24. Only stations 17, 20 and 23 send 'off' each minute when Special Station Auto-Refresh is set.

rayshobby, 2019-01-06:
Judging from the pattern (3 apart), it's likely due to the delay in sending HTTP commands. The relevant code is here:
https://github.com/OpenSprinkler/OpenSprinkler-Firmware/blob/master/OpenSprinkler.cpp#L1040
special stations (such as HTTP) are refreshed one after another with 1 second apart. In switch_httpstation:
https://github.com/OpenSprinkler/OpenSprinkler-Firmware/blob/master/OpenSprinkler.cpp#L1573
it waits for your HTTP server to respond. If the response speed is slow, the next time it comes around to refresh, it may be 3 seconds past the previous one, so it skips the 3 special stations in between, which may explain the symptom you see. There are several possible solutions. One is to remove the code in switch_httpstation where it awaits the server to respond, so it merely sends out the command and is not blocked by the response. Another is to set your HTTP stations 3 stations apart: for example, instead of using 17, 18, 19, 20 and so on, you use 17, 20, 23, 26 and so on. Because HTTP stations are virtual stations and do not require expanders to exist, you can enable as many zones as you want.

bretskee, 2019-01-10:
The other thing I could do is to fix my dumb slow (no) responses. Thanks for the tips though. Working now...

closed 2019-01-10 by bretskee