OpenSprinkler/OpenSprinkler-Firmware #84: Time Zone can't be changed

PipeDeveloper, 2019-04-16:
Hello,

First all, already buy an OpenSprinkler Pi (the extension board for raspberry pi 3) and im using a RPI 3 with a fresh 2019-04-08-raspbian-stretch. full updated.

As the title i just downloaded this firmware as the [tutorial](https://openthings.freshdesk.com/support/solutions/articles/5000631599-installing-and-updating-the-unified-firmware) and configured a basic schedule to test the software. Already im very happy with the performance and a very friendly interface BUT... defining the time by time zone (already checked with the time of the Raspberry pi wich is America/Santiago Chile) GMT / UTC -4 i couldn't change it with the actual time.
[image]
[image]
Which is clearly a difference of 1 hour
Even trying to change the code and compiling again it shows the same diference of one hour.

I would like to change the code that the Time Zone adjust to the local machine time (wich is updated by the system). Also i think that the small battery on the OpenSprinkler Pi board is made to keep the clock running if the pi got disconected.

cmpsci373, 2019-04-24:
The firmware obtains time zone using the location you set in settings. It's independent of the system time zone. If you want to manually set the time zone, clean the Location setting and the time zone will become editable.

rayshobby, 2019-04-24:
The firmware obtains time zone using the location you set in settings. It's independent of the system time zone. If you want to manually set the time zone, clean the Location setting and the time zone will become editable.

closed 2019-04-24 by rayshobby

PipeDeveloper, 2019-04-24:
Hello and thank you for answer. Already i re-installed using a new formated SD card but by default, first run, it set Boston in location field and i couldnt clean it, it ask me to put a location. Even if a change to my location it changes automatically the time zone to -3:00 wich is wrong because it is -4:00

And how the weather function works if i leave that field empty?
Please help I need to change it!!

rayshobby, 2019-04-24:
On the right of the location field there is a cross (x) which you can click to clear the location. Did you try that? Once the location is cleared the time zone field will show up. But the weather feature will not work if there is no set location. The bigger question is why it didn't get your time zone correctly. I will have to look into the time zone script using you specific location. I wonder if there has been any change due to daylight savings time?

PipeDeveloper, 2019-04-24:
Thank you!
I cloud change my time zone but as you said the weather feature is no longer active due empty field on the location.

I think that the script is not showing the time zone correctly maybe for been outdated (many years ago) by the page where it gets the data.

You can check it using the command `sudo raspi-config` and enter to change the clock in "America", Santiago(Chile). or `sudo dpkg-reconfigure tzdata` wich is very useful to automatically set the hour perfectly (no matter summer and winter changes). Chile have only one time zone and by default it is set to -4:00 GTM/UTC.

Have a look into your script and the page wich it gets the data, i think there is the problem, and maybe a manual input to add or take +/-1 would be helpfully even for those who lives in zones wich they change constantly for winter or summer hours and the page doesnt update the time (not my case) but its a good solution at my opinion. Thank you and i really appreciate it

craigcarps, 2019-04-25:
I had a similar challenge when I updated to the latest firmware. To fix it I removed the location setting in open sprinkler settings, manually changed the date / time to the correct value, and then redefined the location. 

This then gave me the correct date/ time, and the location which provided weather inputs.

I suspect there is a quirk in the code where by changing the location doesn’t update the date / time settings anymore in the latest firmware.

rayshobby, 2019-04-27:
Just so you know, I've identified the issue: it's because the timezone package on the weather server is outdated. We will update the related packages shortly and this issue should be fixed very soon.

PipeDeveloper, 2019-04-27:
Thank you very much Ray!