OpenSprinkler/OpenSprinkler-Firmware #87: Monitoring water tank with ultrasound sensor

PipeDeveloper, 2019-05-10:
Hello,
On my irrigation system at home i would like to monitor a water tank of 5800L.

Just short story, my home receive a turn of water specially for irrigation (big yards) with low preasure and sometimes the turn comes at any hour even at 3times per day and sometimes turn doesnt give water in 2-3days, so it's very much better store the water on a tank and irrigate with a pump, that is the system that actually i have.

My old timer just burned. Im still waiting for my OpenSprinkler board wich i plan to use it with a Raspberry Pi 3, already tested this firmware and im very impressed with the features and hope this works.

As the title i would like to monitor the water level of the water tank with an ultrasound sensor, specifically HC-SR04, just a cheap aurduino sensor for reemplacement for any case (probably maintenance on wrong lectures or oxidation for water exposition), obviusly i need to search for a method to protect the sensor inside the tank.

The OpenSprinkler board offer a ways to access to free pins of the raspberry, so its possible to connect the sensor, but i need a little help to add the code to a forked version of the firmware.
I would like to start know wich files i should modificate and also to add it on the webpage (wich i have some expresience with html5, css and js).

My spectation is to collect data in real time of the watter tank, like every 5minutes get 1 data and see on the main page the water level like 40%, 70%, 100%, and on a story line wich i can monitor and know when the tank fills, or see manual irrigation, automated irrigation, etc.
Also in a future define a condition like if the tank is bellow 15% do not irrigate (otherwise when it reach to 0% the pump will suck air).

Thats all, thank you!

rayshobby, 2019-06-20:
We have another product, OpenGarage, which is also based on ESP8266 and has a built-in ultrasonic distance sensor (HC-SR04). It can be used to monitor water tank level.

closed 2019-06-20 by rayshobby

PipeDeveloper, 2019-06-24:
Hello Ray,
I was studing about using the OpenGarage to monitor the water tank level but the thing is the water tank is far way from wifi signal (about 100mts) where i think the most easier way is to wire underground a phone cable (4lines) with the HC-SR04 sensor on the top of the tank, and use the opensprinkler free pins to power the sensor.
Even if i use OpenGarage, the main idea is to visualize the water level on the same page and app of OpenSprinkler.

With your permission,  would like to modify the script of the page and app to add a new field. I think that all is done because i can use as base the OpenGarage App and webpage to extract just a fragment of the distance and the history graphic, just change the titles.
I can use the UI of OpenSprinkler to work on the WebPage
https://openthings.freshdesk.com/support/solutions/articles/5000164006-using-a-different-server-for-ui-assets
But is missing the UI of OpenGarage, i checked the reposory https://github.com/OpenGarage/OpenGarage-Firmware
And i can add the script to read the sensor using the OpenGarage Firmware already im making some testing.

Thank You!