OpenSprinkler/OpenSprinkler-Firmware #96: Allow GPIO pins to be manually entered

dmacias72, 2019-07-17:
I only need a few stations triggered by relays. I compiled  2.1.8 OS 3.0 unified firmware on my own esp8266 with a PCF8574.  I had edited the defines.h, added the address of the PCF8574 and moved the pin definitions to it.  Also created an array of my free pins for the usable free pins on the esp8266. The problem is the app gives me a js error when i click on GPIO. I believe because I'm not running ospi or 2.3. So the freePins var is not defined. So right now I set them through api calls and everything works great. I don't know if the pins could be pulled from the firmware or just allow manual entry.

rayshobby, 2019-10-10:
I don't recommend making this change for OS 3.x, because OS 3.x has pretty much no spare pin. You can certainly create a freePin list, but the issue is that the firmware will still use these pins for things like sensors, etc, even if you don't actually need any sensor. To allow free pins would require commenting out large chunks of the firmware code that operates those pins. I would recommend staying with OSPi and OS2.3 if you want GPIO stations.

rayshobby, 2020-05-24:
Yes the list of configurable pins could be passed from the firmware to the UI, instead of separately defining them in both the firmware and UI. This can be considered for future firmware changes.

closed 2021-02-03 by rayshobby