OpenSprinkler/OpenSprinkler-Firmware #98: Detecting system reboot reason

ruimarinho, 2019-08-12:
Hi,

Occasionally my OpenSprinkler 3.0 reboots with no apparent reason. It could very well due to power spikes but it could also mean a software crash or a software-triggered reboot.

Is there a way to store the last reboot reason to at least get some more debugging information?

eljuanchete, 2019-08-12:
Hello,
I have tried without success to get help on troubleshooting the same issue from Ray's support site.
According to my tests I think that the problem is directly related to two problems: a defective power supply and/or the Wifi signal strength.
When I detected the problem the controller rebooted every few hours. The two problems I detected were:
1.- The refurbished power supply I got (after a replacement of the original one) was defective. This power supply was one of the causes for more frequent reboots. Once I used another power supply (12V, 2A) the reboots were less frequent. However the controller kept rebooting randomly (several hours, at most around 1-2 days).
2.- Next tests were related to Wifi signal strength (one of my first suspicions). If I place the controller near the AP it never reboots randomly (10 days without  reboot). The weaker the signal the more frequent reboots happen. 
I have made many tests and in my case these two seem to be the cause. 
I have checked the firmware (although I am not a programmer) and have seen several places where the controller is rebooted but still cannot confirm if this is really the cause. As far as I know the controller has no USB connector to serial debug the firmware and could not find instructions on how to do it. Neither did I get help on this.
Have you tested if it can be due to the power supply or Wifi signal strength?
It would be great if we can come to a conclusion and get real help from Ray. I am open to make all the tests needed to troubleshoot this.
Regards

nelsonblaha, 2019-08-12:
Are opensprinklers being shipped with faulty power supplies? I have had [similar problems with mine](https://old.reddit.com/r/opensprinkler/comments/ci7bos/kernel_panic_overheat/) freezing and becoming unresponsive, to which others suggested it could be the power supply. I'd love to try a different one to rule it out. Does anyone know a way to buy a power supply with the correct plug?

rayshobby, 2019-08-12:
@eljuanchete: your replacement controller has been tested for over 3 days with the exact power supply shipped to you. There was not a single reboot that I could observe. I don't really know how to diagnose the problem if it cannot be reproduced. If this was a common problem we would have received more reports and would certainly try to address it.

eljuanchete, 2019-08-12:
@nelsonblaha I received a faulty controller and sent it back. The repaired one I received came with a faulty/refurbished power supply. I cannot tell if the original one was faulty. I have no reason to think it was. What I have checked is that the replacement power supply was faulty.
@rayshobby If you cannot reproduce I can. I am sharing my tests and results and am willing to help in anything I can with the troubleshooting. Maybe, as last restart time is not shown on main page many people don't notice it and you don't receive more reports. In fact, I only started to pay attention when I started monitoring the controller uptime.
Can we please troubleshoot it?

rayshobby, 2019-08-12:
There is obviously no reason for us to send you anything that we know are faulty: we paid for all the shipping cost, we didn't ask you for any repair cost,  so why would we go through the trouble just to send you something is knowlingly not to work? I just decided you are reporting something that we are incapable of solving. The controller and power adapter are both tested for 3 days and we know they work, yet when you use them they fail. It's beyond my knowledge now.

rayshobby, 2019-08-12:
Also I know it didn't reboot because during testing we using the test mode feature so it can log onto our own WiFi network. If it rebooted it would have quit and gone into the normal mode that would try to log onto your WiFi which diesn't exist in our testing environment. After over 3 days it stayed on our testing WiFi which means it didn't reboot.  I also don't understand why you said the replacement power adapter is faulty. It's like nothing we send you ever works. So I gave up. Please find another product that fits your need. Both you and I have better things to do with our times.

nelsonblaha, 2019-08-12:
I'm sorry I added my related issue here. Probably my power at my rural home is not stable enough or something. I'll try a UPC.

eljuanchete, 2019-08-12:
I know it is like you say. There is no reason for you to send faulty devices. I didn't say that. But I have double checked and confirmed that the power supply (a refurbished one) is faulty. However, as I said, system reboots not only happen due to power supply but also due to weak WiFi signal and some other process performed due to that poor WiFi signal.
I don't say it didn't reboot on your tests. As we said reboots are unexpected and random and much more frequent with lower strength Wifi signals. I say it reboots on my installation, where many other ESP devices perform without problems and unexpected reboots.
I am only reporting what does not work and asking for help to troubleshoot it.

eljuanchete, 2019-08-12:
In fact, today I moved the controller to a zone where wifi signal is lower and it has rebooted twice since then. Yesterday the controller, very close to the router, was up for 10 days.

rayshobby, 2019-08-12:
If the WiFi signal is low and the controller loses connection, it should reboot: it's a feature, not a bug. The whole point is that if connection is lost such as due to power break it needs a way to reestablish connection.

eljuanchete, 2019-08-12:
Why not try to reconnect instead of rebooting? 
What will happen if controller reboots when a watering program is running? In my case, with latch valves, if the controller does not end the program, valves will keep open and watering continue?

rayshobby, 2019-08-12:
In fact it is trying to reconnect and not reboot. You can take a look at the code to find out. The truth is I don't know why it's rebooting on your end and I can't reproduce it. Why not just find another product that suits your need? I don't understand the obsession with this problem.

rayshobby, 2019-08-12:
The firmware us written in such a way that any reboot happens only after a program is finishes. So the reboot you are observing is likely due to something completely unrelated to firmware. Its maybe the instability of power or something else.

rayshobby, 2019-08-12:
This discussion is going too far away from the firmware discussion and does not belong here anymore so I am closing it now.

closed 2019-08-12 by rayshobby

eljuanchete, 2019-08-12:
My main problem is my last question. something I asked several times in the past. What happens if the controller reboots in the middle of a running program? And also, my problem too is that if it reboots something is not running fine and I cannot see that behavior in any of my other devices so there has to be another problem.

rayshobby, 2019-08-12:
I am sorry our product doesn't fit your need. Please, find another product that satisfies your need.

eljuanchete, 2019-08-12:
The product fits my needs. I am just asking for help and information

ruimarinho, 2019-08-24:
@rayshobby I am sorry if this issue brought up others of which I am unaware of. 

I was simply referring to the fact that other software such as [espurna](https://github.com/xoseperez/espurna/blob/21423431ceab1a6d792d3f6ba746b3166ec6cb5b/code/espurna/crash.ino) save some crash/reboot information. This would be useful in situations like mine where I am trying to diagnose a potential reason for rebooting (could be my VAC adapter, lack of WiFi strength or some other reason).

Would that be something you could consider adding?

rayshobby, 2019-08-24:
The reboot reason has already been added to the upcoming firmware 2.1.9:
https://github.com/OpenSprinkler/OpenSprinkler-Firmware/commit/acbd2cadd0df5729e1736727a338f1a23b020d0c
if the chip rebooted due to crash, there isn't much information to parse. You can connect the controller to a USB serial adapter to check the serial print out, but generally it would just give you an exception code and stack dump, which is obscure information.

rayshobby, 2019-08-24:
I forgot to mention: we are planning to switch to ESP8266 core 2.5.2 (the latest release) in the upcoming firmware 2.1.9:
https://github.com/OpenSprinkler/OpenSprinkler-Firmware/commit/234c0fafbd9062460a48f617b717a1510802e52f
this will hopefully improve WiFi functionality and reduce the chance of crashing.

eljuanchete, 2019-08-24:
If you have the same problem as I did it turned out to be due to WifI disconnections and defective power supply. 
If the controller disconnects for any reason of the WiFi or doesn't reach the gateway after some time (that's what I think) it reboots. You can check it if you connect the controller to a WiFi repeater and then restart the main router. If the router takes some minutes to restart the controller will reboot. You can also check it if you connect the controller near the router (it doesn't reboot  due to Wifi signal loss) and then place it far away from the router. The less strong the WiFi signal, the more chance to disconnect and then probably reboot.
I would also check the power supply. WiFi connection on ESP devices require more power and may cause unexpected reboots if the power supply is not in good state or cannot provide enough current.

eljuanchete, 2019-08-24:
@rayshobby Glad to see you are going to improve the problems I experienced

ruimarinho, 2019-08-25:
That is exactly what I was looking for @rayshobby 👏