OpenSprinkler/OpenSprinkler-Firmware #105: Feature request: Run next scheduled station immediately

stevenfla, 2019-09-28:
If program A is running and we discover perhaps a broken sprinkler head on currently running station 2, we stop this station manually but allow program A to continue. Program A does not start station 3 until later, when it was scheduled to run. 

It would seem useful if station 3 could be started immediately after stopping station 2.

rayshobby, 2019-10-10:
Yes, this has been requested on the forum, and is on our todo list for future firmware.

Nick-Adams-AU, 2020-06-09:
This would be a really handy feature! Any chance we can sneak it into the next release?

rayshobby, 2021-02-03:
This feature has been implemented in a development branch:
https://github.com/OpenSprinkler/OpenSprinkler-Firmware/commit/4479e34f9afc321dd46d28b7729d073cdf276a2c
and we will integrate it into the master branch soon.

closed 2021-02-03 by rayshobby