OpenSprinkler/OpenSprinkler-Firmware #109: Feature Request: Adapt irrigation duration and threshold based on many moisture sensors

jbaudoux, 2019-11-24:
I have several moisture sensors at several depths on several stations. Each sensor station is solar powered and has max 4 sensors. Last sensor is under the roots of the tree and is supposed to not be watered. Others control the progression of the water, ensuring the roots are properly watered and indicates when next water cycle should occur. Data is send in wifi to a mqtt broker and collected in Grafana for plotting. I'll publish an how-to when I'm satisfied with the hardware.

I'm planning to integrate this with opensprinkler to automatically set if the station must run and adapt the watering time.
I thought making the computation outside opensprinkler and send in MQTT to opensprinkler the station duration adjustment and enabling/disabling.

Any advice/remark would be welcome before I start the PR.

tigercjn, 2019-11-25:
We are working on solar sensors as well for our community garden, based on esp8266. They send their message via ESP-Now to a receiver esp8266 connected directly to a Rasberry Pi. The Rasberry does the calculations and send MQTT out via the internet and lights a LED on our map for manual watering/checking. We are also hoping to integrate this with Opensprinkler to run a station as required and the Pi will upload to a database on our website.

leiweke, 2020-06-16:
Hello @jbaudoux @tigercjn I would be very interested to know more about your sensor solution. could you please share some more informations. Many Thanks Jemba

tigercjn, 2020-06-17:
We are very slow but we will get there. 
[image]
It uses capacitance soil measuring.

tigercjn, 2020-06-17:
A custom mounting bracket to hold it and a temperature sensor, it attaches to a 25mm conduit with the wires running inside up to the ESP8266 in a solar-powered enclosure. 
[image]

leiweke, 2020-06-17:
@tigercjn many thanks looks very promising looking forward for more :-)

xaque208, 2020-06-19:
I have similar ambitions.  I've been collecting sensor data using a prometheus exporter that reads data from mqtt, then using grafana to give some insight.  I have the same sort of collection from weather alerts and such, which allows me to define pretty specifically things I care about, like certain temperature thresholds, so I can cover plants on cold nights and such, or move my citrus into the garage.  But I'd love to have some of those soil depth sensors, and also some integration with opensprinkler.  That would be rad.

AlexH0661, 2020-10-29:
Hi all,
I really like the concept of having remote sensor stations, that feedback soil moisture and temperature data, which can then be used to schedule (or pause the schedule) for the zone in which it resides.

Is anyone able to provide a concept diagram of their setups? Doesn't need to be super specific, just conceptually how it works and ties in with OpenSprinkler

jbaudoux, 2020-10-29:
@AlexH0661 Here is a quick first upload to github of my setup:
https://github.com/jbaudoux/moisture

AlexH0661, 2020-10-29:
@jbaudoux , thanks for linking that :) I'll take a look through it today

V1pr, 2023-05-09:
@jbaudoux hi, did you can any closer to your goals? :)