OpenSprinkler/OpenSprinkler-Firmware #110: Feature request: Time of year watering restrictions

mpfl, 2019-12-07:
In Western Australia we [are not permitted to use scheme water to water our gardens during winter](https://www.watercorporation.com.au/save-water/watering-days/winter-sprinkler-ban). It would be great to have the ability to automatically disable watering at certain times of year.

alco28, 2020-02-02:
this should be nice!

craigcarps, 2020-02-02:
I’d also be keen to see this enhancement

SkullKill, 2020-04-19:
+1

linking this
https://opensprinkler.com/forums/topic/enable-disable-programs-via-a-range-of-dates/

rayshobby, 2022-12-13:
This feature has been implemented in the upcoming firmware 2.2.0: for each program, it allows you to set a 'date range' (start date and end date) so that the program is only enabled during the date range.

closed 2022-12-13 by rayshobby