OpenSprinkler/OpenSprinkler-Firmware #112: OpenSprinkler::apply_all_station_bits() issues with Auto Refresh

vk2tds, 2020-01-15:
At the end of the function OpenSprinkler::apply_all_station_bits() there is code that runs if IOPT_SPE_AUTO_REFRESH is set. This code is intended to run once per second, and then send a refresh instruction to each station in turn as required. 

This applies to special stations, such has RF stations, but also to HTTP stations. The issue here is that the HTTP request may take more than one second to run, and because the code does not multi-task, and the station to refresh is based on the time [now()], multiple stations may never actually be refreshed.

Importantly for me, I am using HTTP connected to relays with a watchdog timer. If the relay does not explicitly get refreshed, it will be switched off automatically.

I have developed a fix that will move between the stations sequentially. The code uses the same number of static variables, and works to minimise additional memory use. The modified code is below, and is at the end of the function OpenSprinkler::apply_all_station_bits()

sid is initialised as (MAX_NUM_STATIONS) >>1 so that on startup the refresh starts about half way down the sprinkler zones. This should ensure minimum latency during start up for most people as higher up station ID's are less likely to be populated than lower ones. 

```
        if(iopts[IOPT_SPE_AUTO_REFRESH]) {
                // handle refresh of RF and remote stations
                // we refresh the station whose index is the current time modulo MAX_NUM_STATIONS
                static byte sid = MAX_NUM_STATIONS >> 1;
                static byte lastnow = 0;
                byte now = (now() & 0xff);
                if (lastnow != now){ // refresh once per second only 
                        lastnow = now;
                        sid = (sid + 1) % MAX_NUM_STATIONS;
                        bid=sid>>3;
                        s=sid&0x07;
                        switch_special_station(sid, (station_bits[bid]>>s)&0x01);
                }
        }
```

rayshobby, 2020-05-24:
Ah, good suggestion, will integrate it into the firmware.

rayshobby, 2021-02-03:
A variant of this solution has been integrated in dev-219-7:
https://github.com/OpenSprinkler/OpenSprinkler-Firmware/commit/d8c1bc0987183c4cf7f598fad28944ec53400444

closed 2021-02-03 by rayshobby