OpenSprinkler/OpenSprinkler-Firmware #115: Idea: Better form groups instead of "sequentiell"

opensprinklershop, 2020-04-01:
If you have a lot of stations but limited water, so you want to group your stations.
Example:
I have 4 circles for watering the lawn. I would like to let two of them run in parallel at the desired time (for 30 minutes).

Start program 1 with two circles at 05:00 a.m. - 05:30 a.m.
Start program 2 with two circles at 5:30 a.m. - 6:00 a.m.

I also use the weather control Evapotranspiration (ET). Should this e.g. calculate an irrigation of 150%, then the watering time of 30 minutes would increase. This would result in program one not being finished at 5:30 a.m. but program two already starting. But my water supply would not pack 4 circles. Is there a way that program 2 with the two circles only starts when program 1 is finished, even if the irrigation time has been extended due to the weather adjustment?

A possible solution could be a group number for each station. Only stations with same group nr could run together.

KE-RO, 2020-05-19:
Great idea. An implementation would be great.

rayshobby, 2020-05-24:
If two programs overlap, then the behavior of zone runs depend on the 'sequential' attribute of each zone. By default, all zones are 'sequential', so if two programs overlap, then the zones in the second program will be queued at the end of the first program. The only exception is that if the second program contains the same set of zones -- for example, if zone 5 is running or waiting to run due to program 1, then program 2 has also started, and also requesting zone 5 to run, then the second request will be ignored since zone 5 cannot simultaneously handle two requests.

If zones are 'parallel' (i.e. sequential flag is off), then when the second program starts, zones are allowed to run in parallel so the two programs can overlap in time.

rayshobby, 2020-08-09:
'Sequential group' is being planned in firmware 2.2.0 and is currently in progress in this branch:
https://github.com/OpenSprinkler/OpenSprinkler-Firmware/tree/scheduler-feature

closed 2021-02-03 by rayshobby