OpenSprinkler/OpenSprinkler-Firmware #120: Wrong API Data for ETo

vialkoje, 2020-05-02:
I am using open sprinkler 2.1.9 with ETO configured (2.84mm per day and 646 m elevation ). I suppose that the watering adjustments were ok the last days. When requesting the configuration via API though, the ETo parameters look totally different. Is this a Bug ?

{"devt":1588407834,"nbrd":1,"en":1,"rd":0,"rs":0,"rdst":0,"loc":"48.30311,8.42180","wtkey":"","sunrise":366,"sunset":1242,"eip":1574148195,"lwc":1588406238,"lswc":1588406238,"lupt":1588058010,"lrun":[2,99,238,1588365149],"curr":0,"sbits":[0,0],"ps":[[0,0,0],[0,0,0],[0,0,0],[0,0,0],[0,0,0],[0,0,0],[0,0,0],[0,0,0]],"wto":{"baseETo":0.11,"elevation":600,"pws":""},"ifkey":"","RSSI":31}

vialkoje, 2020-05-03:
I think I found the root-cause. It seems to be just a units issue (mm<>inch meter<>feet) is that by design? Is there a chance to change units for the api ?

rayshobby, 2020-05-24:
Indeed as you can see from the "wto" variable, there is no unit, so by default those numbers are assumed to be in units of inches and feet. We can certainly change it to use metric units but in any case, this is not a firmware issue because the firmware just blindly saves these numbers from the UI and passes them to the weather server, the firmware never uses these numbers itself.

closed 2020-05-24 by rayshobby