OpenSprinkler/OpenSprinkler-Firmware #122: HTTP inaccessible

jslight90, 2020-05-17:
I have been having issues with my new OpenSprinkler 3.2 DC where the HTTP port becomes inaccessible, but I can still ping the IP address on my WiFi.  The only way I am able to regain access is by power cycling the OS.  This issue occurs with both firmware 2.1.8(6) and 2.1.9(3).  The 2.1.9(3) firmware will sometimes reboot itself when the issue occurs, but this causes all running stations to stop with no logs of how long it was running for.

rayshobby, 2020-05-24:
This is more likely a compatibility issue of the WiFi chip with your WiFi router. I don't think this is a know firmware issue. I cannot reproduce this issue on my end -- I have test OS controllers running for months on my WiFi network and do not experience this issue.

closed 2020-05-24 by rayshobby

jslight90, 2020-07-15:
Confirmed.  I replaced my fairly new (1 year old) Netgear R6220 with a new Ubiquiti UAP-AC-HD and have had a solid connection without reboot for over a month.