OpenSprinkler/OpenSprinkler-Firmware #123: Minimal water level

opensprinklershop, 2020-05-26:
Sometimes the calculated water level is really low, such as 3% or 6%. Water programm is then too short to just run once. So it whould be a good idea to define a minimal water level, e.g. 10%. 
When it's under this value, treat it like 0%

Nick-Adams-AU, 2020-06-04:
This would be a great feature! Another option might be to specify it as a time (e.g. 30 seconds).

rayshobby, 2020-06-09:
The firmware already has a logic:
https://github.com/OpenSprinkler/OpenSprinkler-Firmware/blob/master/main.cpp#L728
which stops the zone from watering if: 1) the water level is below 20% and 2) the net water time is below 10 seconds.

Just because water level is low does not automatically mean the zone should stop watering -- it may be programmed with a long water time to begin with (e.g. 3 hours), and even at 5% watering level, the resulting water time may still be significant (9 minutes in this example).

closed 2020-06-09 by rayshobby

Nick-Adams-AU, 2020-06-10:
> The firmware already has a logic:
> https://github.com/OpenSprinkler/OpenSprinkler-Firmware/blob/master/main.cpp#L728
> which stops the zone from watering if: 1) the water level is below 20% and 2) the net water time is below 10 seconds.

Great to hear! Can we look into exposing that 10 second minimum as a configuration variable? For me, starting and stopping a pump for 10 zones every 20 or 30 seconds is expensive and prematurely wears the pump. In my location, we are currently under 20% watering percentage at the moment and it doesn't make sense to run a zone for less than a few minutes.
 
> Just because water level is low does not automatically mean the zone should stop watering -- it may be programmed with a long water time to begin with (e.g. 3 hours), and even at 5% watering level, the resulting water time may still be significant (9 minutes in this example).

This is a great example and I think that having a configuration option would allow this to still occur.

More power to the people :)