OpenSprinkler/OpenSprinkler-Firmware #124: Device unique identifier

travisghansen, 2020-05-27:
I’ve been working on the home assistant integration for opensprinkler and some sort of unique identifier available via the api would be great. I’m not sure if there’s some sort of serial number available in the hardware or even the mac address if nothing else could do.

Thanks for the consideration!

xaque208, 2020-05-28:
There is some machine ID that you can reference I've seen in some other ESP projects.

travisghansen, 2020-05-28:
Any chance we can get it exposed via the api(s)?

psmedley, 2020-05-28:
It should be possible, opengarage exposes the Mac address via the api

rayshobby, 2020-06-09:
This has been addressed in the upcoming firmware (not officially released yet), so I am closing this issue for now.

closed 2020-06-09 by rayshobby