OpenSprinkler/OpenSprinkler-Firmware #125: OS 3.2: Reboot on ENC28J60 freeze or disconnect

opensprinklershop, 2020-05-27:
`diff --git a/main.cpp b/main.cpp
index 24bbf0b..be40613 100644
--- a/main.cpp
+++ b/main.cpp
@@ -377,6 +377,9 @@ void delete_log(char *name);
 void start_server_ap();
 void start_server_client();
 unsigned long reboot_timer = 0;
+
+#define NET_ENC28J60_CHECK_PERIOD                         5000UL  // 5 sec
+uint32_t prevENCCheckTime = 0;
 #endif

 void handle_web_request(char *p);
@@ -410,7 +413,26 @@ void do_loop()
        static ulong connecting_timeout;
        if (m_server) { // if wired Ethernet
                led_blink_ms = 0;
-               Ethernet.maintain(); // todo: is this necessary?
+
+               uint32_t nowTime = millis();
+               if (NET_ENC28J60_CHECK_PERIOD <= (uint32_t) (nowTime - prevENCCheckTime)) {
+                       Ethernet.maintain();
+                       uint16_t phy = Enc28J60.PhyStatus();
+                       if (phy == 0 || phy == 6) {
+                               Serial.println ("ENC28J60 reinit");
+                               os.lcd.setCursor(0, -1);
+                               os.lcd.print(F("Network error!   "));
+                               os.nvdata.reboot_cause = REBOOT_CAUSE_NETWORK_FAIL;
+                               os.status.safe_reboot = 1;
+                       }
+                       prevENCCheckTime = nowTime;
+
+                       //os.lcd.setCursor(0, -1);
+                       //os.lcd.print(nowTime);
+                       //os.lcd.setCursor(7, -1);
+                       //os.lcd.print(Enc28J60.PhyStatus());
+               }
+
                EthernetClient client = m_server->available();
                if (client) {
                        while (true) {
`

opensprinklershop, 2020-06-18:
And just a little bit better - no check on the first 60s - reset timer on ntp sync - fix negative diffs:

```
diff --git a/main.cpp b/main.cpp
index b240844..fad9b0d 100644
--- a/main.cpp
+++ b/main.cpp
@@ -424,17 +424,23 @@ void do_loop()
        if (m_server) { // if wired Ethernet
                led_blink_ms = 0;

-                if (NET_ENC28J60_CHECK_PERIOD <= (uint32_t) (nowTime - prevENCCheckTime)) {
+                if (NET_ENC28J60_CHECK_PERIOD <= (int32_t) (nowTime - prevENCCheckTime)) {
                         Ethernet.maintain();
-                        uint16_t phy = Enc28J60.PhyStatus();
-                        if (phy == 0 || phy == 6) {
-                                Serial.println ("ENC28J60 error");
-                                os.lcd.setCursor(0, -1);
-                                os.lcd.print(F("Network error!   "));
-                                os.nvdata.reboot_cause = REBOOT_CAUSE_NETWORK_FAIL;
-                                os.status.safe_reboot = 1;
+
+                        if (!prevENCCheckTime) {
+                               prevENCCheckTime = nowTime + 60;
                         }
-                        prevENCCheckTime = nowTime;
+                        else {
+                               uint16_t phy = Enc28J60.PhyStatus();
+                               if (phy == 0 || phy == 6) {
+                                       Serial.println ("ENC28J60 error");
+                                       os.lcd.setCursor(0, -1);
+                                       os.lcd.print(F("Network error!   "));
+                                       os.nvdata.reboot_cause = REBOOT_CAUSE_NETWORK_FAIL;
+                                       os.status.safe_reboot = 1;
+                               }
+                               prevENCCheckTime = nowTime;
+                       }
                 }

@@ -1776,6 +1782,7 @@ void perform_ntp_sync() {
                if (t>0) {
                        setTime(t);
                        RTC.set(t);
+                       prevENCCheckTime = 0;
                        DEBUG_PRINTLN(RTC.get());
                        #if !defined(ESP8266)
                        // if rtc was uninitialized and now it is, restart
```

rayshobby, 2020-08-09:
A large part of this is already integrated into 2.1.9(7).

closed 2020-08-09 by rayshobby