OpenSprinkler/OpenSprinkler-Firmware #126: [Feature Request] IFTTT Events at the Start of Station Events

codahq, 2020-05-29:
If we're going to keep a second system in synchronization with the OS (like Hubitat or SmartThings where it's a little more difficult to poll) it would be required if instead of just events at the close of stations there were opening station events as well.  This would allow adhoc station runs to synchronize to another system.  As it stands now, there is just one event for stations at the end.

rayshobby, 2022-12-15:
We did consider adding this, in fact the code has a todo item about adding IFTTT notification to the station start event:
https://github.com/OpenSprinkler/OpenSprinkler-Firmware/blob/master/main.cpp#L1310
but there is a catch -- because connecting to IFTTT server and sending the request take some time, occasionally even up to 15 seconds, this can affect the actual run time of the zone. For that reason, at the moment it's not added. There is already a 'Program Start' event which sends a notification about when a program starts to run. If this is also done on a per-station basis, I am worried this will result in a large number of notifications and affect the actual run time of each zone.

codahq, 2022-12-16:
I built my own binary with the change.  Otherwise, my home automation platform was always out of sync and unless I polled.  I tried polling initially but it always caused the OS to crash.  There must be a memory leak or something.  

At any rate, I still think this is an important feature.  The OpenGarage sends an event any time it opens and that can generate many events.  It would be configurable with a checkbox so people could turn it off if their station events are two infrequent.

rayshobby, 2022-12-16:
Well, OpenGarage is different in that the notification is sent after the door action has been triggered, plus unlike sprinkler zones, there is no 'water time' and the issue with losing some seconds of water time. Also, I imagine the frequency of door opening/closing won't be that frequent, where as a sprinkler program can run up to 72 zones, and if each zone sends a notification, not only they will lose some seconds of water time, but this is a lot of events. In any case, I am glad you built your own binary because that's exactly why I made everything open-source -- you can customize the firmware in any way you want.

rayshobby, 2025-03-06:
This has been addressed in the newly release firmware 2.2.1, which supports Station Start events for IFTTT.

closed 2025-03-06 by rayshobby