OpenSprinkler/OpenSprinkler-Firmware #135: build.sh

pthomas, 2020-06-09:
Hello, I've been using the opensprinkler with Beaglbone Black for many years, and besides having to reinstall it every couple of years it's been great.

This time when I was installing it I noticed an issue with build.sh. For the "osbo" build it does not have the proper '-lmosquitto' at the end like the others do. Actually does it even need to be it's own elif if it's the same as the default?

Also the script shouldn't be calling sudo. Perhaps this should be a proper makefile where the build 'make' can be done as any user and the 'make install' is done as root or with sudo?

Finally, my last comment is maybe instead of trying to install libmosquitto-dev directly it should just be listed as a dependency in the instructions?

thanks,
Paul

rayshobby, 2020-06-09:
I've just updated build.sh. the script calls sudo because it automatically inserts a script to /etc/init.d/ to auto-start opensprinkler process at reboot. Also, it's not the same default as ospi because there is a maco define that distinguishes the two of them (that macro define is passed to the firmware code so it can compile with the correct pin defines).

many users use the automatic update available in the UI and will not be manually typing in command line, so even if installing libmosquitto is just a one time thing, if we ask users to install it manually it may still break it for a lot of users because they don't read instructions.

closed 2020-06-09 by rayshobby

pthomas, 2020-06-09:
OK thanks, that will fix the issue I saw.

Regarding sudo I get why it needs root access for the install, but some
systems won't even have sudo installed, which was the reason I noticed it.

-Paul