OpenSprinkler/OpenSprinkler-Firmware #139: mqtt: opensprinkler availability="offline"

antro31, 2020-06-17:
Hi all,
Just installed the latest OSPI version to enable MQTT usage (thanks to developpers for this!).
I managed to configure the MQTT service but whatever I do, I have nothing published on mqtt. (Apart from opensprinkler/availability set to "offline")
Do I miss something in MQTT configuration?

PeteBa, 2020-06-17:
It would be useful if you could re-build opensprinkler with diagnostic output enabled so that we can see what is going on. You will need to edit `build.sh` and add `-DENABLE_DEBUG` to the OSPI compilation command as follows:

```
	g++ -o OpenSprinkler -DOSPI -DENABLE_DEBUG main.cpp OpenSprinkler.cpp program.cpp server.cpp utils.cpp weather.cpp gpio.cpp etherport.cpp mqtt.cpp -lpthread -lmosquitto
```
Re-build and run the program from the command line. Post the output here.

rayshobby, 2020-06-17:
Double check if the mqtt configuration has been saved correctly. When you change mqtt settings, the dialog box has a 'Submit' button, but that only submits changes to the UI. You still have to click 'Submit' at the top-right corner of the 'Edit Options' page to submit the change to the firmware. Otherwise the mqtt settings will not be saved.

antro31, 2020-06-17:
Hmm.. Interesting. I disabled it before this try. And now, I cannot re-enable it.
I clicked on enabled / submit (on the dialog), then "submit" on the options. Still not enabled. 
Looks like this does not want to take my inputs (tried with 4 different browsers).

> Double check if the mqtt configuration has been saved correctly. When you change mqtt settings, the dialog box has a 'Submit' button, but that only submits changes to the UI. You still have to click [...]

This is what I did already.

antro31, 2020-06-17:
> Hmm.. Interesting. I disabled it before this try. And now, I cannot re-enable it.
> I clicked on enabled / submit (on the dialog), then "submit" on the options. Still not enabled.
> Looks like this does not want to take my inputs (tried with 4 different browsers).
> Is there a way to manually edit the configuration?

Self Reply. Looks like it must be launched as root user. 
Interesting behavior. 
When launched with -DENABLE_DEBUG, I have more mqtt information than without it and now it is shown "online". Looks like the ENABLE_DEBUG option is enabling something else. 

By the way, where can we find the list of topics available for MQTT?

rayshobby, 2020-06-17:
You should make sure that your firmware is up to date with the current master branch. You can go to the OpenSprinkler-Firmware folder, do
git log
the first record you see should be the first commit message of the master branch:
https://github.com/OpenSprinkler/OpenSprinkler-Firmware/commits/master
if not, you should do a
git pull
to make sure you have the current version of the firwmare.

antro31, 2020-06-17:
This is the case. I am on the latest firmware.

rayshobby, 2020-06-17:
Try this: at the homepage, swipe left to right to open the left-side menu, then Export Configurations to a file. This will dump your configurations to a .json file. Open that json file in a text editor, and search for variable "mqtt": what's the value of it (it should be a string inside a pair of curly brackets).

antro31, 2020-06-17:
Actually it contains what I expected: 

`"mqtt":{"en":1,"host":"192.168.1.90","port":1883,"user":"mqtt_user","pass":"mqtt_passwd"}`

But what I wanted to point out is that this behaves differently depending where it is compiled with -DENABLE_DEBUG, or nor.

rayshobby, 2020-06-17:
You mean the "mqtt" string is different depending on whether it's compiled with DENBLE_DEBUG or not?

antro31, 2020-06-17:
No, I mean, when DENABLE_DEBUG is set, this publishes correctly the MQTT information, whereas when not set, it only publishes "availability = offline"

PeteBa, 2020-06-17:
@antro31 , are you able to edit code ?

If so then could you open `mqtt.cpp` and go to line 193. Can you move the line `#if defined(ENABLE_DEBUG)` below the following line `	int state = _loop();` and then test again.

antro31, 2020-06-17:
🥇 Perfect. This works!
Do you plan to be able to respond to MQTT set commands? (enable/disable programs or things like that) ?

rayshobby, 2020-06-24:
Respond to MQTT set command: yes, this is what we've been working at in the mqtt-api branch. It hasn't been integrated yet:
https://github.com/OpenSprinkler/OpenSprinkler-Firmware/tree/mqtt-api

closed 2020-06-24 by rayshobby