OpenSprinkler/OpenSprinkler-Firmware #143: API timestamps are off by the timezone difference

Nick-Adams-AU, 2020-06-26:
Hi,

I'm running firmware 2.1.9(4) on OS hardware version 3.2 DC. I am located in GMT+10 and this has been correctly set in the web app and is correctly returned by the API (`tz` value from `/jo`).

While developing a Home Assistant plugin for OpenSprinkler, the devs have [noticed](https://github.com/vinteo/hass-opensprinkler/issues/89#issuecomment-649937085) that timestamps returned by the OS API seem to be incorrect by exactly the timezone offset.

Here's an example:

Actual/correct timestamp: `1593144102` which is `06/26/2020 @ 4:01am (UTC)` or `06/26/2020 @ 2:01pm (GMT+10)`
Incorrect "devt" timestamp: `1593180102` which is `06/26/2020 @ 2:01pm (UTC)`

You can see that the timestamp is out by exactly the timezone difference (10 * 3600 = 36000). This error also seems to impact the last run zone time ("lrun") and possibly other timestamps.

The OS web app seems to display the correct time however it doesn't display the timezone so I suspect it is just "failing correctly".

rayshobby, 2020-06-26:
The 'devt' variable is always the 'local time'. This has been explained in the API document.

closed 2020-06-26 by rayshobby

Nick-Adams-AU, 2020-06-27:
It's a bug then as the `devt` shows a timestamp from the future and not the local time.

A`timestamp` is a fixed point in time (seconds since 1/1/1970 at **UTC**) and its value is the same for every location, irrespective of their timezone. The timestamp being presented by OS is not the current time - it is ahead by the timezone offset.

Nick-Adams-AU, 2020-06-27:
Please see this comment:

https://github.com/vinteo/hass-opensprinkler/issues/89#issuecomment-650327394

`now()` is already the correct timestamp...without any adjustment for timezone. To put it in simpler terms, now() is already now!

rayshobby, 2020-06-27:
I don't think you understood what devt represents. Look, devt returns now_tz(), not now()
https://github.com/OpenSprinkler/OpenSprinkler-Firmware/blob/master/server.cpp#L1109
this is NOT a bug, it's intentional by design. now() represents UTC+0 time, and now_tz() (i.e. devt) represents UTC shifted by your local time zone. You are telling me what you want devt to me, but you are not trying to understand what devt actually represents. Nobody said timestamp must be UTC+0 time.

Whatever scripting language you use, you can always read devt and convert it to whatever format you want.

Nick-Adams-AU, 2020-06-27:
>  Nobody said timestamp must be UTC+0 time.

With respect, your documentation says "Device time (epoch time)" and "epoch time" is clearly defined here:

https://en.wikipedia.org/wiki/Unix_time

...as `the Unix epoch is 00:00:00 UTC on 1 January 1970.`. It would be a reasonable expectation that a timestamp would be a regular unix timestamp that basically every library understands and can process without pre-processing.

Another example, the `local time`, `timestamp` or `unix timestamp` for UTC, GMT+10, GMT-10 and GMT+5 all resolve to the same value. The timezone doesn't matter. 

If people want to show a human-friendly time in a GUI, they should use the API provided timezone or even better, use the app specific timezone to convert the device provided UTC value.

rayshobby, 2020-06-27:
Sure, I will modify the document to clarify it then. Look, I explained what devt is, and I think the explanation is unambigious, so I don't really know what it is that you want then.

rayshobby, 2020-06-27:
There is a reason why devt is defined this way: my assumption is that when you look at devt, no matter where you are to access the controller, it always represents the device's local time zone time, regardless of whether you are accessing the controller in the US, or Europe, or Asia. That's why devt includes the time zone shift in it. Otherwise, to know the device's local time, you would also have to look up for the time zone information.

Nick-Adams-AU, 2020-06-27:
Your assumption is still correct and very valid and you should continue this as the intended behaviour. The problem is with the current OS implementation.

The current unix timestamp or epoch is the same no matter what timezone you are in. Here is the current unix timestamp/epoch as I compose this:

UTC = 1593219365
GMT+10 = 1593219365
GMT-10 = 1593219365
GMT+1  = 1593219365

The problem arises in that the timestamp OS is presenting is in the future (now() + offset) which as you saw above, will be in the future...it hasn't happened.

The correct way to handle this is to present `devt` as now() and let the web app or API consumer display the time using any timezone offset they wish - as per your example - US, Europe, Asia. The same will go for any other timestamps.

now() is now everywhere in the world...no matter what timezone you are in.

rayshobby, 2020-06-27:
If you want UTC+0 time (i.e. time-zone independent), why do you even need to read devt then?? Any scripting language can already give you this, it's globally the same as you said. There is no point querying devt for that -- you can already get it without even accessing OpenSprinkler. How about this: just think of devt as carrying time zone information (i.e. devt - epoch represents the device's time zone)? Then you can forget about devt being epoch time or not.

Nick-Adams-AU, 2020-06-27:
I use `devt` as a simple example of the issue but the same problem exists with `lrun` and other timestamps returned by the API. Currently, my `lrun` value is in the future!

People might still use `devt` to make sure the OS device time matches whatever system time they are processing on. For example, I might want to show when a zone starts but my app might be a few seconds ahead or behind `devt` so I'll align them when I present it to the user.

>  How about this: just think of devt as carrying time zone information (i.e. devt - epoch represents the device's time zone)? Then you can forget about devt being epoch time or not.

timestamps don't carry timezone information! That's the problem.

> As I said, you are still trying to tell me what you want devt to be, while ignoring how we use it.

I'm trying to bring OS into line with standard, accepted programming practices so other developers can easily use your product. Time is hard...I've made so many bugs with time that I've learnt my lessons. [It's not just me that thinks this practice is unusual](https://github.com/vinteo/hass-opensprinkler/issues/89#issuecomment-650464045).

You can still keep your desired goal of localised time display without having to force any API consumers to kludge for misrepresented time. IMHO, your documentation is actually correct and the implementation should kindly be updated to bring it into line with standard behaviour rather than continue with some redefinition of time.

rayshobby, 2020-06-27:
devt is NOT timestamp, how about that?

I already said: you can just understand devt as time zone, forget about epoch time, it's time zone, ok?

So, to convert lrun to the format you want, just do (lrun + current_epoch_time - devt). That's it.

Look, this conversation is not going anywhere. We can't change devt at the moment because the entire UI/app is built on what devt means. Just to say it again, it's not a timestamp, how about that?

Nick-Adams-AU, 2020-06-27:
> We can't change devt at the moment because the entire UI/app is built on what devt means.

If this is just a resource/difficulty thing, would you accept a PR that fixed this or am I wasting my time?

> Again, devt is not what you think it is, we will change the document to clarify it.

My apologies. I was just going by the documentation. Thanks for offering to clarify it!

rayshobby, 2020-06-27:
The UI is designed to work with all previous versions of firmware, so if you change it, users on previous firmwares will all get the times displayed incorrectly. So we can't change it. You can create a new API variable, say 'timestamp', and return 'now()' as its value. Or, since I already explained what 'devt' represents, the simplest solution is to just use:
devt - 3600/4*(json.options['tz']-48)
to counteract the timezone offset, and get the timestamp that you want. Here json.options['tz'] is variable 'tz's value.