OpenSprinkler/OpenSprinkler-Firmware #146: Feature Request: Increase the number of Fixed Start Times per Program

trbolexis, 2020-07-16:
I have some specific use cases where more than 4 unique start times per program would be ideal (instead of using the repeating function).

I tried increasing the [code]MAX_NUM_STARTIMES[/code] in program.h to 10 (from 4), compiled the sketch, and uploaded it to my OS2.3...

[image]

... it broke my ability to save any program. Beyond that, in the web-front end, only 4 fixed start time options were visible.

If possible, could someone point me in the correct direction to expand this functionality? With the structure of the programming, I'm not 100% sure where to begin with my modifications.

Thanks.

trbolexis, 2020-07-17:
I was able to figure this out (mostly) .. The key was in identifying that the UI is .js and hosted online.

[image]

[image]

Next up, I need to figure out how to get the updated "App" online and have the firmware point to it.

rayshobby, 2020-07-19:
Please note that you can always duplicate the program and modify the start times. That way you get 4 additional start times. If you need 12 start times, you can duplicate the program twice. When using a large number of start times, generally it's on a regular interval so you can use the Repeating option. A large number of start times while not on regular interval is not very common.

rayshobby, 2020-08-09:
I am closing this issue for now. There is no plan to increase the number of fixed start times beyond 4 at the moment. If you need more than 4, the first thing to look is to see if you can split the program start times into sub-groups of regular intervals; then you can always duplicate the program as many times as you need, each will give you another 4 unique start times.

closed 2020-08-09 by rayshobby

trbolexis, 2020-08-09:
OK.

I have modified the source code, and added 1 additional Master Valve, and
10 total fixed start-times.

I am not too familiar with Git, can someone point me in the direction to
create a new branch for Opensprinkler development? I'd be glad to share my
code.