OpenSprinkler/OpenSprinkler-Firmware #147: HTTP station receives "off" command repeatedly

xaque208, 2020-07-18:
Hello, I'm wondering if this is by design, or perhaps if there is a way to disable this.  I've noticed today that an HTTP station continues to receive whatever is in the "off" field of the HTTP station, and this appears to happen for each HTTP station I have.

Is there a way to disable this behavior I'm overlooking?  This becomes a problem when using the osbee as a remote http station, I'm unable to get above a minute of runtime due to the off command being called repeatedly.  I don't remember this being an issue last year, but now I'm wondering how long this has been happening.  Any guidance here would be appreciated.

Running 2.1.9 rev 4

rayshobby, 2020-07-19:
Go to Edit Options -> Advanced -> uncheck "Special Station Auto Refresh".

closed 2020-07-19 by rayshobby

xaque208, 2020-07-25:
Amazing, thank you!