OpenSprinkler/OpenSprinkler-Firmware #156: dependency on 'wiringpi'

ChrisDumont, 2020-12-04:
In gpio.cpp line 266 there is a call to 'gpio'. /usr/bin/gpio is supplied by the 'wiringpi' package (which wasn't on my system). I fairly quickly discovered this but it could be troublesome for some users. There's no error checking in the code regarding if the call succeeds or fails; perhaps there should be.
This dependency could be either:
- mentioned in the user manual
- installed by 'build.sh' similarly to libmosquitto-dev

rayshobby, 2021-02-03:
It used to be that wiringpi is installed by default in raspbian, but it may be different now. I will add it to build.sh as you said.

closed 2021-02-03 by rayshobby

varac, 2022-06-29:
See also #196 about the current state of wiringpi in RPi OS bullsey