OpenSprinkler/OpenSprinkler-Firmware #159: Relatively high CPU usage on RPI 2B

outsideset, 2021-01-05:
I noticed relatively high cpu usage on my rpi 2b. OpenSprinkler firmware runs at about 9%, continuously.

I tried increasing the value for delay() on main.cpp:1006, but that didn't work. Search results regarding delay() vs sleep() indicate:

> Sleep relinquishes the CPU for specified amount , so that other process can execute. While Delay is just busy waiting for the amount to expire and then continue once expires, it do not execute any other process during this period.

I tried changing it to sleep and had better results. sleep(1) gets me about 7% CPU, and sleep(10) makes CPU usage intermittent and less than ~3%.

Is there a reason delay() is being used as opposed to sleep()? Any reason not to increase the sleep to 10ms?

outsideset, 2021-01-05:
looks like the UI is affected by using sleep. sleep(1) the UI is a little sluggish... sleep(10) and its very sluggish. So its not so easy to fix I guess... I'm guessing the arduino combability must be part of the story.

rayshobby, 2021-02-03:
My recollection is that delay() did reduce CPU usage, that's why we inserted that in the program. If it didn't reduce CPU usage, we wouldn't have left delay there. Perhaps the way delay is implement in the Rpi system is different now? But in any case, I would think less than 10% CPU usage is acceptable.

closed 2021-02-04 by rayshobby