OpenSprinkler/OpenSprinkler-Firmware #160: API request - Get all PIDs

AlexH0661, 2021-01-12:
I'm reading through the latest API documentation at the moment, but it doesn't look like there is a way to get the PID associated with each program. What is the most appropriate method to get the PID of all programs on the OpenSprinkler controller **programatically** (API, MQTT, etc)?

Running a query against `/jp` is meant to return:
```
**Return Variables:**
● nprogs: Number of existing programs.
● nboards: Number of 8-zone groups (including main controller).
● mnp: Maximum number of programs allowed (40 in this firmware).
● mnst: Maximum number of program start times (fixed time type) allowed (4 in this firmware).
● pnsize: Maximum number of characters allowed in program name (32 in this firmware).
● pd: Array of program data. Each element corresponds to a program. See below for data structure.
```
Which seems to include everything, except the PID of each program.

I am currently writing a python script which makes use of MQTT and API calls to achieve some additional automation, but part of the program initially enumerates all programs on the OpenSprinkler controller and will make use of the PIDs (as they are required for every subsequent API call that modifies programs).

AlexH0661, 2021-01-12:
@jbaudoux , do you have any ideas?

rayshobby, 2021-02-03:
The PID (i.e. program index) is implied in the order the programs are stored. In the program array, if you see the array is like:
["summer program", "winter program", "monday program" ...]
then the PID of them are 0, 1, 2... in that order. For that reason, the index is not stored in the program. What you are asking is akin to asking "what's the PID of the **second** program" -- that's already implied in the question itself.

Perhaps what you are asking is to find the index using the program name?

AlexH0661, 2021-02-03:
Hi Ray,

That makes sense.

I believe you are correct. I'm writing my program with the assumption that it knows very limited if not nothing about the controller. Part of the program enumerates the controller to determine if it's program already exists, and if it doesn't it creates it.

I will add logic that enumerates the list of programs and looks for the name of it's program. If it doesn't find it, it will then create a new program using ID that is generated by counting the number of programs.

Thanks for your response.

Kind regards,
Alex

closed 2021-02-03 by AlexH0661