OpenSprinkler/OpenSprinkler-Firmware #162: Code uncompilable: Ethernet.tick() and readRegister are not publically accessible

maxgerhardt, 2021-02-05:
The code states

https://github.com/OpenSprinkler/OpenSprinkler-Firmware/blob/0a86f6c5a9b865cbf6ae058c0c8816f446def085/OpenSprinkler.cpp#L2350-L2355

As if `Ethernet.tick()` was a publically callable function on the Ethernet object.

However, the library (and I assumed you used https://github.com/UIPEthernet/UIPEthernet for the ESP8266) states 

https://github.com/UIPEthernet/UIPEthernet/blob/a44baee6e990737abb04877887cf43db4fab73c5/UIPEthernet.h#L109-L124

Thus the compilation error 

```
.pio\libdeps\d1_mini_lite\UIPEthernet/UIPEthernet.h:116:15: error: 'static void UIPEthernetClass::tick()' is private
   static void tick();
               ^
OpenSprinkler.cpp:2351:24: error: within this context
  if(m_server) Ethernet.tick();
                        ^
In file included from OpenSprinkler.h:38:0,
                 from OpenSprinkler.cpp:24:
```

so.. either you're using an entirely different Ethernet library for the ESP8266 or the code doesn't compile for you as well? 

Same for these calls

https://github.com/OpenSprinkler/OpenSprinkler-Firmware/blob/0a86f6c5a9b865cbf6ae058c0c8816f446def085/main.cpp#L446-L448

`readReg` is private within the class

```
In file included from .pio\libdeps\d1_mini_lite\UIPEthernet/UIPEthernet.h:35:0,
                 from OpenSprinkler.h:38,
                 from main.cpp:26:
.pio\libdeps\d1_mini_lite\UIPEthernet/utility/Enc28J60Network.h:246:18: error: 'static uint8_t Enc28J60Network::readReg(uint8_t)' is private
   static uint8_t readReg(uint8_t address);
                  ^
main.cpp:446:31: error: within this context
     uint16_t estat = Enc28J60.readReg((uint8_t) ENC28J60_ESTAT);
                               ^
```

On a side-note, the PR I submited for the PlatformIO build got broken since the library author for the ESP8266/ESP32 OLED library decided to delete the library from the registry or rename it -- a fixed `platformio.ini` is attached

(must be used with the fix to comment out `Ethernet.tick()` and comment out `#define ENABLE_DEBUG` in `defines.h`)

```
[platformio]
src_dir = .
include_dir = .

[env:d1_mini_lite]
platform = espressif8266
board = d1_mini_lite
framework = arduino
lib_ldf_mode = deep
lib_deps =
     uipethernet/UIPEthernet @ 2.0.8
     sui77/rc-switch @ ^2.6.3
     https://github.com/ThingPulse/esp8266-oled-ssd1306/archive/4.2.0.zip
     knolleary/PubSubClient @ ^2.8
```
I'l address this in a future PR as well as the build for the AVR.

rayshobby, 2021-02-05:
Please use this modified version of UIPEthernet:
https://github.com/OpenSprinkler/UIPEthernet/branches
in the fixes/dhcp branch. It contains improvements and fixes to the dhcp code.

maxgerhardt, 2021-02-05:
Thanks, this instantly made the code builtable when using this version of library. 

```
Checking size .pio\build\d1_mini_lite\firmware.elf
Building .pio\build\d1_mini_lite\firmware.bin
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [=====     ]  48.4% (used 39628 bytes from 81920 bytes)
Flash: [=====     ]  45.8% (used 438968 bytes from 958448 bytes)
Creating BIN file ".pio\build\d1_mini_lite\firmware.bin" using "C:\Users\Max\.platformio\packages\framework-arduinoespressif8266\bootloaders\eboot\eboot.elf" and ".pio\build\d1_mini_lite\firmware.elf"
============================================= [SUCCESS] Took 13.59 seconds =============================================
```
PR'ing new `platformio.ini` soon.

closed 2021-02-05 by maxgerhardt