OpenSprinkler/OpenSprinkler-Firmware #163: Accessing private FatFile* FatVolume::vwd() function in SDFat

maxgerhardt, 2021-02-05:
When attempting to compile this firmware for The ATMega1284P Sangduino, with the config 

```
[env:sanguino_atmega1284p]
platform = atmelavr
board = ATmega1284P
board_build.f_cpu = 16000000L
board_build.variant = sanguino
framework = arduino
lib_ldf_mode = deep
lib_deps =
     UIPEthernet=https://github.com/OpenSprinkler/UIPEthernet/archive/fixes/dhcp.zip
     knolleary/PubSubClient @ ^2.8
     greiman/SdFat @ ^2.0.4
     Wire
```

Since that's what seems to be used in 

https://github.com/OpenSprinkler/OpenSprinkler-Firmware/blob/443865d4b8d84c49b8949f99e7150ea770ab5e9c/make.os23#L8-L12

The following compilation error occurrs

```
.pio\libdeps\sanguino_atmega1284p\SdFat\src/FatLib/FatVolume.h: In function 'void delete_log(char*)':
.pio\libdeps\sanguino_atmega1284p\SdFat\src/FatLib/FatVolume.h:336:12: error: 'FatFile* FatVolume::vwd()' is private
   FatFile* vwd() {return &m_vwd;}
            ^
main.cpp:1679:11: error: within this context
    sd.vwd()->rmRfStar();
           ^
```

again the `main.cpp` does a `sd.vwd()` call but that function is `private`. 

https://github.com/greiman/SdFat/blob/cb5a7d2afb4eb94f2563044645fc816f0ea4c2c8/src/FatLib/FatVolume.h#L333-L336

I did not see a modified version of the SDFat library in https://github.com/rayshobby?tab=repositories&q=&type=&language= or https://github.com/OpenSprinkler, so what library is being used here exactly?

rayshobby, 2021-02-06:
The firmware was compiled with SdFat version 1.0.7. I didn't realize the latest version has changed this function to private. Will address it later.

maxgerhardt, 2021-02-06:
Well thanks if I use 1.0.7 it works. I additionally had to ignore the soure code in the `html` folder but now it compiles both AVR + ESP8266 firmwares :)

I'll create a draft PR that works for now and if you change the library version, that can be easily changed in the `platformio.ini`.

StohanzlMart, 2021-02-17:
My new favourite action movie,
the saints doing wonders!

Thank you!

rayshobby, 2022-11-04:
The platformio.ini is now included in the firmware code folder, and it specifies the correct version of SdFat library to use. The firmware code can probably be adapted to the latest version of SdFat library (by using the rmdir function), but I will leave it as is for now.

closed 2022-11-04 by rayshobby