OpenSprinkler/OpenSprinkler-Firmware #168: Expose BSSID when in AP mode

ruimarinho, 2021-04-04:
Hi @rayshobby,

It would be great to display the BSSID of the network found when displaying the list of networks under AP mode. Currently, each AP for the same network name is displayed alongside a signal strength, but one has to deduce the correct AP according to its signal.

This assuming it is guaranteed that opensprinkler will use that specific AP, which in my case I don't think was the case.

Thanks,

rayshobby, 2022-11-04:
This should be relatively easy to implement. It looks like ESP8266 core library does allow connection to a specific bssid if it's specified. I will put this on my todo list (though I don't have a mesh network so I will probably have to find someone to help test it).

psmedley, 2022-11-04:
I can test this, I have a mesh network and an os3.0 at home.

rayshobby, 2022-11-22:
We've released a test version of firmware 2.2.0(0), per forum post:
https://opensprinkler.com/forums/topic/first-version-of-firmware-2-2-0-ready-for-testing/

It has added support for saving BSSID and channel information. The related commit history is here:
https://github.com/OpenSprinkler/OpenSprinkler-Firmware/commit/a5279a37e0cfc68880776fe21a91f864af7df04b

You can try it out to test. Thanks.

closed 2022-12-13 by rayshobby