OpenSprinkler/OpenSprinkler-Firmware #175: how to pull other git builds on rpi?

docsparks, 2021-04-29:
sorry to ask here, but after a frustrating amount of tunnel searching, i cant help but ask.

i use two different RPIs , and follow the instructions here 
[https://openthings.freshdesk.com/support/solutions/articles/5000631599](url)

and it pulls like
git clone https://github.com/OpenSprinkler/OpenSprinkler-Firmware.git

how could I change this to (for example)
https://github.com/rianadon/OpenSprinkler-Firmware/tree/scheduler-extracompat

changing to 
git clone https://github.com/rianadon/OpenSprinkler-Firmware/tree/scheduler-extracompat/OpenSprinkler-Firmware.git
does not work.   

any advice would be greatly appreciated, thank you!

rianadon, 2021-04-29:
No problem! I'd recommend cloning to another directory and copy your data files over into another folder like `OpenSprinklerFirmware-v2`. For example:

```bash
git clone https://github.com/rianadon/OpenSprinkler-Firmware OpenSprinklerFirmware-v2
cp OpenSprinklerFirmware/*.dat OpenSprinklerFirmware-v2/
cd OpenSprinklerFirmware-v2
git checkout scheduler-extracompat
sudo ./build.sh ospi
```

That way you have a backup of your `.dat` files just in case, and you can copy them back to the main folder when things get merged.

You could alternatively add a remote to where you've cloned the git repository (`git remote add...`) and then checkout the branch from the new remote, but I find the method above less of a pain.

docsparks, 2021-05-05:
about to install this and give this thing a go!   hate to sound needy, but any chances you could have the web-front end display the list of zones into two columns?   or maybe a column for every 8 zones?  i have a display always on and scrolling down and all over the place is annoying. 40 some odd zones, id love to have them shown side by side.

if not, no big deal, there is always homeasssistant thank god, but you seem like youd say ez-pz to the task, again thank you!

will post feedback when i get to the testing of the git update

rianadon, 2021-05-05:
I like the idea. If you want something hacky, you can fork [my fork](https://github.com/rianadon/OS-App) and add these lines under `docs/css/app.css`:
```css
@media (min-width: 1032px) {
    #os-running-stations, #os-stations-list { columns: 2; }
}
```
If GitHub doesn't already do it for you after forking, you'll need to [set up GItHub pages](https://guides.github.com/features/pages/) to use the `docs/` folder as the source, then you can substitute my frontend url for yours.

This code snippet doesn't look great when stations are running, so I don't want to commit this. But I'll create an issue on app repository and maybe someone there will have a better solution. Nevertheless, this sounds like the perfect problem to be solved by a home assistant dashboard.

But feel free to create future issues related to my stuff on my forked repository. That way we don't congest the issue tracker on this repo with off-topic things.

closed 2021-05-06 by docsparks