OpenSprinkler/OpenSprinkler-Firmware #176: Sequential and non-sequential zones at the same time

Heimchen, 2021-05-03:
Hi,
if I add sequential zones and non-sequential zones to a program, the sequential zones and the first non-sequential zone will be executed in a sequence, but the following non-sequential zones will be started at the same time like the first sequential zone.
I would expect that all sequential zones are executed sequentially, and afterwards all non-sequential zones are started in parallel.
Regards

rayshobby, 2021-05-21:
This is expected behavior of non-sequential zones. If you want one non-sequential zone to run after another non-sequential zone, that's effectively a separate sequential group, that is, you want them to be serialized. Serial groups have been implemented in a new branch but haven't been merged into the master branch yet:
https://github.com/OpenSprinkler/OpenSprinkler-Firmware/tree/scheduler-feature

Heimchen, 2021-05-23:
I would expect that sequential zones will never run at the same time as non-sequential zones. Isn't that correct? All non-sequential zones may run at the same time. So if zones 1+2+3 are sequential and zones 4+5+6 are non-sequential and all zones shall run for 30min, I would expect that zone1 runs for 30min, afterwards zone 2, afterwards zone 3 and afterwards zones 4+5+6 together.

rayshobby, 2021-05-25:
No -- if a group of zones are sequential, that means they are guaranteed to be serialized. However, there is no guarantee that a non-sequential zone cannot run with them -- that's the definition of non-sequential, that is, they can run with other zones. If you are looking to modify this behavior, feel free to modify the firmware code in any way you want.

closed 2021-05-25 by rayshobby