OpenSprinkler/OpenSprinkler-Firmware #180: 2.1.9(7) release?

henfri, 2021-05-25:
Hello,

under releases, I find only releases up to 2.1.9(4), but in the forum, I see 2.1.9(7) bing the latest version.
Could you please also release a 2.1.9(7?)
Also, I think that to update to this version, a binary would be needed. Where are these bing published?

Best regards,
Hendrik

rayshobby, 2021-05-25:
Just pull from the master branch. Please take a look at OSPi firmware instructions:
https://openthings.freshdesk.com/support/solutions/articles/5000631599-installing-and-updating-the-unified-firmware-on-ospi
we no longer create releases for firmwares -- the master branch is always the most current firmware.

closed 2021-05-25 by rayshobby

henfri, 2021-05-25:
Hello,

thanks for letting me know. What a pitty. 
I can with some effort compile myself, but how does it work for all the other users of opensprinkler?

Anyway: It would maybe be good to have this as some kind of final note under releases.

Best regards,
Hendrik

rayshobby, 2021-05-26:
I wasn't sure if you are looking for the firmware for OSPi or for OS 3. For OSPi or any Linux-based platform, there is no pre-compiled binary, because the binary can't be shared among the different platforms (i.e. RPi, Linux, BeagleBone do not use the same binaries), and compiling it is very easy so there is no need to provide binary. For OS 3, the pre-compiled binary files are in this github repo:
https://github.com/OpenSprinkler/OpenSprinkler-Compiled-Firmware

henfri, 2021-05-26:
Hi Ray,

understood.

I still think, that a new "release" under https://github.com/OpenSprinkler/OpenSprinkler-Firmware/releases would be good, pointing towards the new location - as I think most would search there.
But it's up to you, of course.

Best regards,
Hendrik