OpenSprinkler/OpenSprinkler-Firmware #182: Development on Scheduler UI Branch

trbolexis, 2021-06-03:
Is this branch of Opensprinkler with the group scheduling feature functional? Is it actively being developed?

I am planning to continue development, and need an idea if this will be a useful place from which to start my work.

rianadon, 2021-06-10:
If you do mean the `scheduler-feature` branch, I've built on it in #173 which would be a good place to continue from.

rayshobby, 2022-11-08:
Most of the changes in the scheduler-features branch are now merged into firmware 2.2.0 source code:
https://github.com/OpenSprinkler/OpenSprinkler-Firmware/tree/dev/os220
I am currently testing and fixing bugs if I find any. The scheduler UI branch will also be gradually merged. I am closing this ticket now since it's being actively developed. Once it's released, if there are any issues you can re-open the ticket.

closed 2022-11-08 by rayshobby