OpenSprinkler/OpenSprinkler-Firmware #184: Proposal: duration with MQTT station state 1

lukepalmer, 2021-06-17:
Today, MQTT messages for when stations go on and off look like this:

opensprinkler/station/0 {"state":1}
opensprinkler/station/0 {"state":0,"duration":6}

I think the duration should be included in the first message. 
Why?
Suppose I want to use these messages to control a remote station. What happens if:

- remote station sees the state:1 message and turns on
- my MQTT broker blows up
- water runs forever

It seems better to include a duration in the state:1 message, as a way of saying "I expect you to run for this long".

If Ray and others like this I'm happy to make a PR.

rayshobby, 2022-11-04:
The reason it's included when the zone goes off is that the user can turn off the zone at any time, so the actual run time is only known accurately when the zone is turned off. Of course this is assuming the MQTT message is mainly used for monitoring / record keeping of zone runs, we didn't think about the application scenario that you are using these messages to turn on / off a remote station. This information can certainly be added when a zone turns on as the 'expected duration'. Will put this on the todo list.

rayshobby, 2022-11-08:
This has been added in the firmware 2.2.0 source code;
https://github.com/OpenSprinkler/OpenSprinkler-Firmware/blob/dev/os220/main.cpp#L1044
I am closing this ticket for now.

closed 2022-11-08 by rayshobby