OpenSprinkler/OpenSprinkler-Firmware #185: Help: Increase Size of Station Attributes Data

trbolexis, 2021-06-26:
I have modified the firmware code base (and app code base) to add additional master stations -> 8 Master Stations.

Everything works perfectly in operation, but the individual station attributes do no persist after a reboot. For example:

If I disable SXX and reboot the OS3.0 the station will be enabled again.
If I set a ignore rain delay, and ignore sensor 2 on SXX the options are then deselected after reboot.

I get the feeling that the memory space for the station attributes is a specific size, and the new data is not saving correctly to SPIFFS (or wherever), and not restoring on power up.

Can someone point me in the correct direction to remedy this? Thank you.

[image]
[image]
[image]

docsparks, 2021-08-06:
did you resolve this?  any chances to share where you changed?

rayshobby, 2022-11-04:
Can you post the changes you made somewhere? It's difficult to reason about the issue without seeing the code.

closed 2022-12-13 by rayshobby