OpenSprinkler/OpenSprinkler-Firmware #186: ESP8266 OS 3.X Compile Error

PipeDeveloper, 2021-07-27:
Following this guide, I have installed 2.7.4 and 3.0.2 core and both have the same issue
https://openthings.freshdesk.com/support/solutions/articles/5000165132-how-to-compile-opensprinkler-firmware

There is something related with the pubsubclient library
https://github.com/knolleary/pubsubclient

After the command `make -f make.lin32`
I got this output

```
flash_hal.cpp
core_esp8266_app_entry_noextra4k.cpp
/root/esp8266_2.7.4/libraries/pubsubclient/tests/src/receive_spec.cpp: In function 'int test_receive_stream()':
/root/esp8266_2.7.4/libraries/pubsubclient/tests/src/receive_spec.cpp:65:12: error: cannot declare variable 'stream' to be of abstract type 'Stream'
     Stream stream;
            ^
In file included from /root/esp8266_2.7.4//cores/esp8266/HardwareSerial.h:32:0,
                 from /root/esp8266_2.7.4//cores/esp8266/Arduino.h:245,
                 from /root/esp8266_2.7.4//libraries/pubsubclient/src/PubSubClient.h:10,
                 from /root/esp8266_2.7.4/libraries/pubsubclient/tests/src/receive_spec.cpp:1:
/root/esp8266_2.7.4//cores/esp8266/Stream.h:38:7: note:   because the following virtual functions are pure within 'Stream':
 class Stream: public Print {
       ^
In file included from /root/esp8266_2.7.4//cores/esp8266/Stream.h:26:0,
                 from /root/esp8266_2.7.4//cores/esp8266/HardwareSerial.h:32,
                 from /root/esp8266_2.7.4//cores/esp8266/Arduino.h:245,
                 from /root/esp8266_2.7.4//libraries/pubsubclient/src/PubSubClient.h:10,
                 from /root/esp8266_2.7.4/libraries/pubsubclient/tests/src/receive_spec.cpp:1:
/root/esp8266_2.7.4//cores/esp8266/Print.h:57:24: note:         virtual size_t Print::write(uint8_t)
         virtual size_t write(uint8_t) = 0;
                        ^
In file included from /root/esp8266_2.7.4//cores/esp8266/HardwareSerial.h:32:0,
                 from /root/esp8266_2.7.4//cores/esp8266/Arduino.h:245,
                 from /root/esp8266_2.7.4//libraries/pubsubclient/src/PubSubClient.h:10,
                 from /root/esp8266_2.7.4/libraries/pubsubclient/tests/src/receive_spec.cpp:1:
/root/esp8266_2.7.4//cores/esp8266/Stream.h:47:21: note:        virtual int Stream::available()
         virtual int available() = 0;
                     ^
/root/esp8266_2.7.4//cores/esp8266/Stream.h:48:21: note:        virtual int Stream::read()
         virtual int read() = 0;
                     ^
/root/esp8266_2.7.4//cores/esp8266/Stream.h:49:21: note:        virtual int Stream::peek()
         virtual int peek() = 0;
                     ^
/root/esp8266_2.7.4/libraries/pubsubclient/tests/src/receive_spec.cpp:66:12: error: 'class Stream' has no member named 'expect'
     stream.expect((uint8_t*)"payload",7);
            ^
In file included from /root/esp8266_2.7.4/libraries/pubsubclient/tests/src/receive_spec.cpp:4:0:
/root/esp8266_2.7.4/libraries/pubsubclient/tests/src/receive_spec.cpp:89:21: error: 'class Stream' has no member named 'error'
     IS_FALSE(stream.error());
                     ^
/root/esp8266_2.7.4//libraries/pubsubclient/tests/src/lib/BDDTest.h:11:62: note: in definition of macro 'TEST'
 #define TEST(x) { if (!bddtest_test(__FILE__, __LINE__, #x, (x))) return false;  }
                                                              ^
/root/esp8266_2.7.4/libraries/pubsubclient/tests/src/receive_spec.cpp:89:5: note: in expansion of macro 'IS_FALSE'
     IS_FALSE(stream.error());
     ^
/root/esp8266_2.7.4/libraries/pubsubclient/tests/src/receive_spec.cpp: In function 'int test_receive_oversized_stream_message()':
/root/esp8266_2.7.4/libraries/pubsubclient/tests/src/receive_spec.cpp:253:12: error: cannot declare variable 'stream' to be of abstract type 'Stream'
     Stream stream;
            ^
In file included from /root/esp8266_2.7.4//cores/esp8266/HardwareSerial.h:32:0,
                 from /root/esp8266_2.7.4//cores/esp8266/Arduino.h:245,
                 from /root/esp8266_2.7.4//libraries/pubsubclient/src/PubSubClient.h:10,
                 from /root/esp8266_2.7.4/libraries/pubsubclient/tests/src/receive_spec.cpp:1:
/root/esp8266_2.7.4//cores/esp8266/Stream.h:38:7: note:   since type 'Stream' has pure virtual functions
 class Stream: public Print {
       ^
/root/esp8266_2.7.4/libraries/pubsubclient/tests/src/receive_spec.cpp:276:12: error: 'class Stream' has no member named 'expect'
     stream.expect(bigPublish+9,length-9);
            ^
In file included from /root/esp8266_2.7.4/libraries/pubsubclient/tests/src/receive_spec.cpp:4:0:
/root/esp8266_2.7.4/libraries/pubsubclient/tests/src/receive_spec.cpp:287:21: error: 'class Stream' has no member named 'error'
     IS_FALSE(stream.error());
                     ^
/root/esp8266_2.7.4//libraries/pubsubclient/tests/src/lib/BDDTest.h:11:62: note: in definition of macro 'TEST'
 #define TEST(x) { if (!bddtest_test(__FILE__, __LINE__, #x, (x))) return false;  }
                                                              ^
/root/esp8266_2.7.4/libraries/pubsubclient/tests/src/receive_spec.cpp:287:5: note: in expansion of macro 'IS_FALSE'
     IS_FALSE(stream.error());
     ^
makeEspArduino.mk:257: recipe for target '/tmp/mainArduino/mainArduino_generic/receive_spec.cpp.o' failed
make: *** [/tmp/mainArduino/mainArduino_generic/receive_spec.cpp.o] Error 1
make: *** Waiting for unfinished jobs....

```

PipeDeveloper, 2021-07-27:
I made a VM with ubuntu 20.04, clean OS.
I installed the latest version of the ESP8266 core and the latest version of OpenSprinkler Firmware

Same output:

```
flash_hal.cpp
core_esp8266_app_entry_noextra4k.cpp
/root/Arduino/libraries/pubsubclient/tests/src/receive_spec.cpp: In function 'int test_receive_stream()':
/root/Arduino/libraries/pubsubclient/tests/src/receive_spec.cpp:65:12: error: cannot declare variable 'stream' to be of abstract type 'Stream'
   65 |     Stream stream;
      |            ^~~~~~
In file included from /root/Arduino//cores/esp8266/HardwareSerial.h:32,
                 from /root/Arduino//cores/esp8266/Arduino.h:288,
                 from /root/Arduino//libraries/pubsubclient/src/PubSubClient.h:10,
                 from /root/Arduino/libraries/pubsubclient/tests/src/receive_spec.cpp:1:
/root/Arduino//cores/esp8266/Stream.h:50:7: note:   because the following virtual functions are pure within 'Stream':
   50 | class Stream: public Print {
      |       ^~~~~~
In file included from /root/Arduino//cores/esp8266/Stream.h:27,
                 from /root/Arduino//cores/esp8266/HardwareSerial.h:32,
                 from /root/Arduino//cores/esp8266/Arduino.h:288,
                 from /root/Arduino//libraries/pubsubclient/src/PubSubClient.h:10,
                 from /root/Arduino/libraries/pubsubclient/tests/src/receive_spec.cpp:1:
/root/Arduino//cores/esp8266/Print.h:55:24: note:     'virtual size_t Print::write(uint8_t)'
   55 |         virtual size_t write(uint8_t) = 0;
      |                        ^~~~~
In file included from /root/Arduino//cores/esp8266/HardwareSerial.h:32,
                 from /root/Arduino//cores/esp8266/Arduino.h:288,
                 from /root/Arduino//libraries/pubsubclient/src/PubSubClient.h:10,
                 from /root/Arduino/libraries/pubsubclient/tests/src/receive_spec.cpp:1:
/root/Arduino//cores/esp8266/Stream.h:59:21: note:     'virtual int Stream::available()'
   59 |         virtual int available() = 0;
      |                     ^~~~~~~~~
/root/Arduino//cores/esp8266/Stream.h:60:21: note:     'virtual int Stream::read()'
   60 |         virtual int read() = 0;
      |                     ^~~~
/root/Arduino//cores/esp8266/Stream.h:61:21: note:     'virtual int Stream::peek()'
   61 |         virtual int peek() = 0;
      |                     ^~~~
/root/Arduino/libraries/pubsubclient/tests/src/receive_spec.cpp:66:12: error: 'class Stream' has no member named 'expect'
   66 |     stream.expect((uint8_t*)"payload",7);
      |            ^~~~~~
In file included from /root/Arduino/libraries/pubsubclient/tests/src/receive_spec.cpp:4:
/root/Arduino/libraries/pubsubclient/tests/src/receive_spec.cpp:89:21: error: 'class Stream' has no member named 'error'
   89 |     IS_FALSE(stream.error());
      |                     ^~~~~
/root/Arduino//libraries/pubsubclient/tests/src/lib/BDDTest.h:11:62: note: in definition of macro 'TEST'
   11 | #define TEST(x) { if (!bddtest_test(__FILE__, __LINE__, #x, (x))) return false;  }
      |                                                              ^
/root/Arduino/libraries/pubsubclient/tests/src/receive_spec.cpp:89:5: note: in expansion of macro 'IS_FALSE'
   89 |     IS_FALSE(stream.error());
      |     ^~~~~~~~
/root/Arduino/libraries/pubsubclient/tests/src/receive_spec.cpp: In function 'int test_receive_oversized_stream_message()':
/root/Arduino/libraries/pubsubclient/tests/src/receive_spec.cpp:253:12: error: cannot declare variable 'stream' to be of abstract type 'Stream'
  253 |     Stream stream;
      |            ^~~~~~
/root/Arduino/libraries/pubsubclient/tests/src/receive_spec.cpp:276:12: error: 'class Stream' has no member named 'expect'
  276 |     stream.expect(bigPublish+9,length-9);
      |            ^~~~~~
In file included from /root/Arduino/libraries/pubsubclient/tests/src/receive_spec.cpp:4:
/root/Arduino/libraries/pubsubclient/tests/src/receive_spec.cpp:287:21: error: 'class Stream' has no member named 'error'
  287 |     IS_FALSE(stream.error());
      |                     ^~~~~
/root/Arduino//libraries/pubsubclient/tests/src/lib/BDDTest.h:11:62: note: in definition of macro 'TEST'
   11 | #define TEST(x) { if (!bddtest_test(__FILE__, __LINE__, #x, (x))) return false;  }
      |                                                              ^
/root/Arduino/libraries/pubsubclient/tests/src/receive_spec.cpp:287:5: note: in expansion of macro 'IS_FALSE'
  287 |     IS_FALSE(stream.error());
      |     ^~~~~~~~
exc-sethandler.cpp
sqrt32.cpp
makeEspArduino.mk:257: recipe for target '/tmp/mainArduino/mainArduino_generic/receive_spec.cpp.o' failed
make: *** [/tmp/mainArduino/mainArduino_generic/receive_spec.cpp.o] Error 1
make: *** Waiting for unfinished jobs....
```

StohanzlMart, 2021-08-07:
Try PlatformIO, just clone this repo and "open project" in PIO. Saves you a lot of hassle including library management.

Worked on my Windows x64 machine without any troubles since the platformio.ini file is already populated.

PipeDeveloper, 2021-09-08:
Thank you. Yes platformIO works like a charm. Thank you.
Closed

closed 2021-09-08 by PipeDeveloper